PX4/PX4-Autopilot #9697: Crash in RC failsafe with vision position

cimbar, 2018-06-16:
To describe issue clearly, i will explain all changes on multicopter with details;

**1-** In my setup, i was testing OFFBOARD flight with vision position and range sensor by using companion computer connected to TELEM2. There wasn't any problem on flight in this setup but after some time has passed TELEM2 started not to work, i couldn't connect to pixhawk via companion computer. I changed both the companion computer, pixhawk hardware (used one) and connection cable but the result was same (I couldn't test new pixhawk hardware properly but i tried). 

**2-** Then, firstly, i have connected the companion computer to USB port and faced stabilization problems in flight with STABILIZED mode. Secondly, i've made a connection between companion computer and TELEM1 by using USB to TTL converter. In this flight, i faced with stabilization problem again.

**3-** At the end of the unstable flight, the UAV has crashed. Because of the swings on orientation, i have cut off the thrust and engaged the manual kill switch. But, it suddenly changed the manual kill switch to off (i didn't do anything on RC) and started to rise. I have lost all connection with vehicle and it crashed to a wall. At least, the battery was unplugged after crash and the motors shut down. You can check the log of last flight.

https://review.px4.io/plot_app?log=8fd0349a-035b-4fd0-93b7-4724c64be5af

I don't have any logical reason for unstable flight, the only change was the connection port and baudrate. According to log, my opinion is the RC connection has lost because of the hard landing when i cut off the thrust (it was close to ground). When the RC connection has lost, failsafe was activated and it tried to land. I suppose it couldn't recognize it was on ground and it has started to fly again because there wasn't GPS connection. at the end it drifted to left and hit the wall.

When failsafe mode is enabled with vision position flight, could there be a bug of safe landing? And how did the vehicle change the manual kill switch to off?

LorenzMeier, 2018-06-16:
You've lost RC input:

[image]

And the log suggests the kill switch was only active very briefly:

[image]

cimbar, 2018-06-16:
Yes, the FCU changed the manual kill switch to 'off' itself and the vehicle started to ascend although it was on the ground after RC lost. While it was ascending, it slided to left and crashed to the wall.

LorenzMeier, 2018-06-17:
No, the FCU doesn't modify the logic itself. It merely was only on for a split second. We are logging the radio inputs separately but at a lower rate and the kill switch was never on long enough to be captured there. I can also see that at around the same time other controls on the remote (like the param tuning channels) changed slightly and returned to their nominal value right after.

From the logs it looks like the kill switch was not put into the "kill" position but flipped back and the system lost RC connection short after (see the log above).

What probably lead to the crash in the first place is the strong imbalance of the output motors. It looks like a motor rotation issue, mounting issue or issue with the center of gravity.

Leading up to the crash the lines diverge more and more, but for an indoor flight they should have been roughly on top of each other.

[image]

closed 2018-06-17 by LorenzMeier