PX4/PX4-Autopilot #9706: Missing airspeed sensors drivers for px4 v5 ?

jstrebel, 2018-06-18:
Hello, 
I try to hook up the airspeedsensor 4525DO  and the SDP3X on the pixhawk4 (px4 v5) without success. 
PX4 software 1.8rc (compiled on 16.6)
In the QGC I get the message “no airspeed sensor found”.
The airspeedsensor 4525 works with the Pixhawk2. 
I have connected an I2C protocol analyzer to the I2C connection to the sensor. On the analyzer I see that the PX4 is sending data.
doing ls /dev, I dont see the  ms4525. 
I guess the drivers is not built for px4 v5. Maybe I am wrong.  Jakob

DanielePettenuzzo, 2018-06-18:
@jstrebel on what port did you connect your sensor? the issue is that the pixhawk 4 has 3 external i2c ports and probably the driver for your airspeed sensor is not starting automatically on the port that you want. I'm currently working on fixing this. For now you can try to start it manually by launching the following command in the mavlink shell:
``` 
sdp3x_airspeed start -b <bus_number>
```
where: 
I2CA is bus_number 4
I2CB is bus_number 2

jstrebel, 2018-06-18:
Daniele, Thank you for the quick response.
I connected the airspeedsensor on the I2C A

DanielePettenuzzo, 2018-06-18:
This port is not checked by default.
Let me know if it works with:
``` 
sdp3x_airspeed start -b 4
```

jstrebel, 2018-06-18:
Daniele, sorry for my ignorance. how does the command for the ms4525 look like?
jakob

DanielePettenuzzo, 2018-06-18:
No problem.
```
ms4525_airspeed start -b 4
```
By the way did you flash master or the release from qgc? The release from qgc doesn't fully support pixhawk 4. This will come with the next release. With master everything seems to be working for me.

jstrebel, 2018-06-18:
Daniele, actually this was my guess. But it does not start. Also not with - b 2

DanielePettenuzzo, 2018-06-18:
What message do you get in the console when it fails to start? 
I just tried starting the ms4525 on I2CA with this command and it works on latest master. 

Other possible issues could be the length of the wires or if one of the wires is not properly connected. I'm using the ms4525 from MRobotics and this has the same pinout as the pixhawk 4.

jstrebel, 2018-06-18:
Daniele, I did flash the beta from saturday and QGC from 6.6. I will be back Friday to let you know the exact message I got. Thanks jakob

jstrebel, 2018-06-22:
@DanielePettenuzzo,
thank you for your help. I tried sdp3x_airspeed start -b 4 and get the following response.
Q: is the message below important or can Ignore it?

Start the individual calibration steps by clicking one of the buttons to the left.
[cal] calibration started: 2 airspeed
[cal] Ensure sensor is not measuring wind
command denied during calibration: 241
command denied during calibration: 241
command denied during calibration: 241
[cal] Offset of 0 Pascal
[cal] Blow across front of pitot without touching
[cal] Create air pressure! (got 0, wanted: 50 Pa)
[cal] Create air pressure! (got 0, wanted: 50 Pa)
[cal] Create air pressure! (got 0, wanted: 50 Pa)
[cal] Create air pressure! (got 0, wanted: 50 Pa)
[cal] Positive pressure: OK (96 Pa)
[cal] calibration done: airspeed

dagar, 2018-07-12:
I believe this has been fixed, please reopen if not.

raygamero, 2018-08-03:
Today I installed the stable release and developer release.  Both did not start the SDP3X driver.  I had to manually start it on my Pixhawk 4.

raygamero, 2018-08-06:
Give me some time to test.  Looking at my parameter set, CBRK_AIRSPD_CHK is set to 0.

achalap, 2018-09-06:
I tried with QGC V3.4.3 with MS4525, still does not work.  I set the CBRK_AIRSPD_CHK to 0.  I am using HILStar(XPlane) airframe.

closed 2019-02-04 by stale[bot]