PX4/PX4-Autopilot #9707: Bypassing libuavcan for Nuttx-CAN interface

Mani-Sadhasivam, 2018-06-18:
Hello,

I have a usecase where I have to use PX4 stack without libuavcan for accessing the Nuttx-CAN interface on Aerocore2 board (based on STM32 microcontroller). Currently I have modelled by CAN module based on the [Nuttx CAN example](https://github.com/PX4/NuttX/blob/master/apps/examples/can/can_main.c). While executing the module I didn't get any error but I can't see the traffic on CAN bus.

Can anyone, @pavel-kirienko provide some input here?

PS: I have used `/dev/can0'  as the CAN device as per Nuttx example.

Mani-Sadhasivam, 2018-06-19:
Reopening since the associated forum topic is not yet approved by the moderator!

Can anyone provide some info here?

bkueng, 2018-06-20:
@davids5 do you know more?

davids5, 2018-06-21:
@Mani-Sadhasivam - you will have to enable the CONFIG_CAN in the nuttx defconfig. See https://github.com/PX4/Firmware/blob/master/src/drivers/boards/aerocore2/can.c#L59-L69

Mani-Sadhasivam, 2018-06-21:
@davids5 I have enabled that option and the write command succeeds but still I can't see any frames on the bus! But with UAVCAN I can see proper CAN frames. Can you see whats going wrong here? I'm sure that the STM32 driver successfully sends the CAN frames and it returns the sent bytes. However the frames are not visible on the CAN bus when probed.

davids5, 2018-06-21:
@Mani-Sadhasivam  - 

Am I correct to assume you have uavcan removed from the build when you are testing this?

Where are you calling `dev_caninit()`? It should be called late in `board_app_initialize`

I would suggest you step through the code and see if the correct bus is being used, and if there are errors reported back by the CAN HW.

Mani-Sadhasivam, 2018-06-21:
I'm calling 'dev_caninit' in my module and using CAN bus 1! I did look into the code and debugged it and everything seems to be correct but still nothing on the CAN bus. Is there any pinmux settings need to be set? Is UAVCAN doing anything special than the Nuttx CAN driver?

There is no error btw from the STM32 CAN driver. It correctly returns the bytes transferred.

davids5, 2018-06-21:
Pin init is done in stm32_caninitialize - check them in your board.h and see if they match your HW.

pavel-kirienko, 2018-06-25:
@Mani-Sadhasivam the symptoms you described could be because of bad connectivity with other nodes on the bus. Remember, a CAN bus can function only if there are at least two nodes on it.

Mani-Sadhasivam, 2018-07-04:
I have nailed down the issue. My CAN sniffer was single-ended, so by mistake I was probing it on CAN-H but it should have done on CAN-L. After correcting this, I can see the frames in CAN bus.

Thanks everyone for your support!

closed 2018-07-04 by Mani-Sadhasivam

jshubham11, 2019-05-13:
Hi Mani, @Mani-Sadhasivam 
I am trying to do the same. Can you please guide me through the steps. I am not able to find the links mentioned in your issue and replies.

I am trying to do the same on Pixhawk 4.