PX4/PX4-Autopilot #9713: Baro spikes in SITL

RomanBapst, 2018-06-19:
For some reason the baro data in SITL (gazebo) is very spiky.
[image]

This causes the ekf height checks to fail. @JohannesBrand FYI

dagar, 2018-06-19:
They're present in the pressure coming straight from the baro simulator driver, so it must be from gazebo.

https://github.com/PX4/sitl_gazebo/blob/10b0597cd20def97074c6c8eade580a7f294e41d/src/gazebo_mavlink_interface.cpp#L509-L516

dagar, 2018-06-19:
@priseborough could you review the gazebo barometer noise?

https://github.com/PX4/sitl_gazebo/blob/10b0597cd20def97074c6c8eade580a7f294e41d/src/gazebo_mavlink_interface.cpp#L518-L532

TSC21, 2018-06-25:
This is also having consequences on LPE: https://github.com/PX4/Firmware/issues/9713

priseborough, 2018-06-25:
rand() returns a integer and x1 and x2 are supposed to be from a uniform random distribution between +- 1.0 so so lines  522 and 523 should be

     x1 = 2.0 * ((double)rand() / (double)RAND_MAX) - 1.0;
     x2 = 2.0 * ((double)rand() / (double)RAND_MAX) - 1.0;

If we had ranf() available to us, that would be a more efficient way of generating x1 and x2.

Use of Gaussian/Normal distributions is common in robotics and flight vehicle simulation, so support for this in the math library would be appropriate.

LorenzMeier, 2018-07-03:
Can you send a PR? Thanks!

priseborough, 2018-07-04:
Fixed here: https://github.com/Dronecode/sitl_gazebo/pull/43

closed 2018-07-13 by priseborough