PX4/PX4-Autopilot #9716: EKF2 Indoor Mag Mode not functional

mhkabir, 2018-06-19:
**Describe the bug**
The EKF2 Indoor Mag fusion mode (EKF2_MAG_TYPE = 4) does not function on master and v1.8 release.

**To Reproduce**
Set EKF2_MAG_TYPE=4 and reboot vehicle, or restart ekf2 in SITL (`gazebo_iris_opt_flow`).

**Expected behavior**
The EKF should initialize and produce an attitude estimate. 

In this case, the EKF does not initialise or produce an attitude estimate.

**Log Files and Screenshots**
Log from SITL : https://logs.px4.io/plot_app?log=5c2e7e20-35de-4278-bee8-40ba1b92c242

```
pxh> ekf2 status
INFO  [ekf2] local position OK no
INFO  [ekf2] global position OK no
INFO  [ekf2] time slip: 0 us
```

priseborough, 2018-06-20:
This fixes it on replay: https://github.com/PX4/ecl/tree/pr-ekfBugFix.

 I also noticed the artificial sensor noise on the SITL simulation which is the same on-ground and in-flight is causing the _vehicle_at_rest check to fail, which turns off fusion of the last known yaw angle, allowing the yaw to drift when on ground. I do not see this on hardware.

 If you don't get a chance, I will test further tomorrow.

mhkabir, 2018-06-20:
Just tested, it seems that yaw angle drift is present on ground. It also drifts quite a lot in the air, which is not entirely unexpected, but I expected better given that I'm using a temperature calibrated gyro.

Here is a replay log : https://review.px4.io/plot_app?log=c3632182-1b19-4c9c-9a19-a2816d6b982c

mhkabir, 2018-06-22:
Whoops, I thought I had logging from boot enabled. Sorry, will redo.

priseborough, 2018-08-24:
@mhkabir ping

closed 2019-02-04 by stale[bot]