PX4/PX4-Autopilot #9726: EKF2 wind estimate very wrong

ryanjAA, 2018-06-20:
Flown on 6/18 on Master (so almost 1.8 stable)

Actual wind at site was between 12-15 m/s

EKF2 Wind estimate shows ~2-4 m/s.

https://review.px4.io/plot_app?log=e6d82fe3-9b3b-4f4b-b8c6-5af1e930e1a9

Also, if you take look at ground speed vs airspeed we're cruising at about 20 m/s but traveling closer to 35 m/s but wind estimates are nowhere near what we were actually in (we measure wind with an anemometer before testing) and looking at surrounding area wind readings all were consistent with ours. It was blowing your shirt against your body kind of wind, not gentle breeze which is more the 2-5 m/s reported.

Master on 6/18
fmuv3 firmware on Original Pixhawk (2mb flash)
QGC Daily also from 6/18

@dagar

dagar, 2018-06-20:
Here's a plot of ground speed, wind speed (+ components).

[image]

[e6d82fe3-9b3b-4f4b-b8c6-5af1e930e1a9.ulg.pdf](https://github.com/PX4/Firmware/files/2120699/e6d82fe3-9b3b-4f4b-b8c6-5af1e930e1a9.ulg.pdf)

dagar, 2018-06-20:
Same plot again with TAS (true airspeed) innovations, variance, and test ratio.

[image]

dagar, 2018-06-20:
When the TAS test ratio (dark red in the plot above) is greater than 1 the airspeed is rejected.

dagar, 2018-06-20:
Do we just need a larger airspeed gate (EKF2_TAS_GATE) for a situation like this with sufficiently high wind?

ryanjAA, 2018-06-28:
What would be the next step here? We fly in wind usually in the 10-20 mph pretty regularly so while not as extreme as this specific log we can changes things for the next flight once understanding what would be ideal.

closed 2019-02-04 by stale[bot]