PX4/PX4-Autopilot #9729: Can't bind a Specktrum receiver to Pixhawk 4 board.

nobbynofriends, 2018-06-20:
Hi guys,

First time posting, I've had a quick look across the group but I can't see a post that answers my issue:

I’ve just bought a pixhawk 4 board from holybros and I can’t get a spektrum dsmx satellite receiver to connect to my transmitter

I’ve tried all three options in the spektrum bind option in QGroundControl, but each time the status led on the GPS module flashes green, there’s three quick beeps, the led on the receiver flashes once and then nothing.  

I've also sucessfully bound the satellite receiver using the master receiver.  But when I plug it into the pixhawk, it doesn't light up and QGround Control says there is no connection.

I’ve tried using an identical satellite receiver and got the same results.  There is 3.29V being supplied from the DSM/S-BUS RC socket and the supplied cable has good continuity.  I've also tried powering the pixhawk from a battery as well as USB.

The satellite receivers are about 7-8 years old, although they do still work.  Could this be an issue?

Transmitter: Spektrum DX8. Tried using 11ms and 22ms frame rate as well as an additional Spectrum DX7.
Receivers: Two satellite receivers from two Spektrum AR8000 receiver kits. One was bound, one had never been used.
Pixhawk 4 board: 1.7.3dev set to generic ground vehicle (I’m building a rover).
QGround Control v3.3.2

Log to follow.

LorenzMeier, 2018-06-21:
Please upgrade the Pixhawk 4 board to v1.8 and also potentially try a different "airframe" vehicle config.

nobbynofriends, 2018-06-23:
Hi Lorenz, I've now tried both of those things and still have exactly the same issue.  Any more suggestions?

nobbynofriends, 2018-06-28:
Hello, this has been solved now.  The Pixhawk - DSM cable supplied with the Pixhawk was incorrectly wired on the receiver end.  Swapping the voltage and signal wires solves the issue.

closed 2018-06-28 by nobbynofriends