PX4/PX4-Autopilot #9735: MC yaw setpoint changing on increasing throttle

CarlosSpin, 2018-06-21:
**Describe the bug**
While performing checks for a Pixhawk first flight, I removed the propellers from the multicopter, armed the controller while indoors, and increased the throttle. Quickly the yaw setpoint starts to increase, although no such command is issued, which causes some motors to stop.

All sensor checks were performed and were successful. The Radio is correctly calibrated.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on without propellers
2. Drone armed
3. Throttle increased

**Expected behavior**
Should be expecting an increase in throttle while the stick is up, followed by a decrease when the stick is released.

**Log Files and Screenshots**
Two separate logs:
https://logs.px4.io/plot_app?log=4a71a025-7490-4955-bf6c-1d82c0e1cd87
https://logs.px4.io/plot_app?log=f75796af-eda9-4e42-b3d4-cf6ba66a5f06

**Drone (please complete the following information):**
- The drone is a 10" coaxial octocopter
- The pixhawk is mounted with the Yaw at 180º, and the sensors are configured as such.

LorenzMeier, 2018-06-22:
You need to increase PWM_MIN in the parameters section to ensure no motors stop. What you see is the yaw integral winding up as a result of the test without propellers. The core issue is that your PWM output range is incorrectly configured.

closed 2018-06-22 by LorenzMeier

CarlosSpin, 2018-06-22:
Thank you for your help.

I tried to set all PID gains (for the yawrate and yaw) to 0, and increased the PWM_MIN value as you suggested (which was 1075 and is now 1125).

I performed another ground test, in the same setup, and verified the same yaw setpoint increase. The yaw rate setpoint is now always 0, as expected since no controller is present. Shouldn't this be the same case with the yaw setpoint?

How can I make sure that when I take off, the vehicle won't immediately try to rotate in yaw, as suggested by the logs?

Thank you