PX4/PX4-Autopilot #9737: FrSky Telemetry Not Recognizing S-PORT

rollysing, 2018-06-21:
I've tried setting up the FrSky telemetry using a Craft & Theory Pixhawk cable per [documentation](https://docs.px4.io/en/peripherals/frsky_telemetry.html). The C&T cable is connected to the S PORT of the X6R and the other end daisy-chained into a FrSky LiPo Voltage sensor then to TELEM2 of the Pixhawk (eBay 2.4.6 version).

I've tried it with both stable and dev versions of 1.6.5, 1.7.0, and 1.7.3. frsky_telemetry status in the Mavlink console says: ERROR [frsky_telemetry] running: DTYPE . As I'm made to understand, it should auto sense SPORT. I've tried swapping the Rx and Tx pins and also connecting TELEM2 directly into Rx S PORT to no avail. I tried 1.8.0 stable and it doesn't even recognize the frsky_telemetry status command.

All of the above setup changes were per @bkueng recommended diagnostic step. Apparently, the only way to force it to switch from DTYPE to SPORT is a change in the source code?

Please advise or recommend resolution.

Thanks.

rollysing, 2018-06-22:
This seem to be related to this: https://github.com/PX4/Firmware/issues/9240

rollysing, 2018-06-22:
From the Slack thread:
I told @bkueng 
Hopefully this could be resolved. Otherwise, I've got $40 worth of useless cables, unless I switch to ArduPilot?

@bkueng Trying with ArduPilot is actually a good idea. Then we know if it's a software or hardware issue.

Me: Yes, it works in ArduPilot. I loaded Mission Planner 1.3.56 and ArduPlane V3.8.5. I only had the GPS connected and it immediately discovered the coordinates. So to confirm, it's not a hardware issue. By the way, ArduPilot uses the SERIAL 4/5 port instead of TELEM2. The SERIAL4_PROTOCOL param is changed from 5 to 10.

bkueng, 2018-07-18:
I recently made some improvements for frsky telemetry. You can now (by flashing master) specify the mode via `frsky_telemetry start -m sport`.

rollysing, 2018-08-05:
@bkueng I'm not a coder or anything so could you please give me a step-by-step on how to do this? Is this something that has to be put in the SD card? Let me know.

Thanks.

rollysing, 2018-08-15:
@hamishwillee My apologies. I'm just seeing this now. I didn't mention it here but I believe I did originally in the Slack forum where this started and @bkueng requested I open an issue. I'm not a coder/programmer. I'm simply a hobbyist/user with what I still consider a "newbie status" so everything you mentioned just went over my head. If you or someone can provide me the file to put in the etc\extras.txt, that I'm capable of.

Many thanks.

hamishwillee, 2018-08-15:
@rollysing I'm not a programmer either. Unfortunately, once you're talking about things that require more than just download and set a parameter, then you need to be willing to read docs and try stuff. 

I guess we should step back. 
1. What flight controller hardware are you using?

@bkueng When will there be firmware that can be downloaded from QGC that has this  - next release?

rollysing, 2018-08-15:
@hamishwillee I've tried it in Pixhawk 1 mRobotics and generic eBay ones v3.4.6 and v3.4.8.

hamishwillee, 2018-08-15:
The fact is that if you want to do this you will almost certainly need to build firmware. This will be non-trivial. Are you willing to try that?

bkueng, 2018-08-16:
> @bkueng When will there be firmware that can be downloaded from QGC that has this - next release?

Yes next release. But you can also flash master via QGC if you select 'Advanced'

rollysing, 2018-08-16:
@bkueng So in the next release, this will be supported natively without having to do the frsky_telemetry start -m sport step as you mentioned earlier as long as the steps to setup are followed according to the [FrSky Telemetry PX4 Docs](https://docs.px4.io/en/peripherals/frsky_telemetry.html)?

Thanks.

bkueng, 2018-08-20:
> @bkueng So in the next release, this will be supported natively without having to do the frsky_telemetry start -m sport step as you mentioned earlier as long as the steps to setup are followed according to the FrSky Telemetry PX4 Docs?

Yes, but this is already the case with the current release. I'm not sure what the problem is with your setup.

rollysing, 2018-08-20:
My mRobotics Pixhawk 1 came in today and tried to setup per docs and your recommendation. No joy! Please look at the setup and let me know what I'm missing.

[image]

[image]

I even tried this but no luck:
[image]

[image]

All I get is RSSI and RxBat (which is even wrong.)
[image]

I appreciate any assistance.

Thanks.

bkueng, 2018-08-21:
> I even tried this but no luck:

Unfortunately `frsky_telemetry` is disabled on the v2 build target (Pixhawk), due to Flash limitations. 
Did you flash it yourself via QGC? If the bootloader is new enough, it should automatically select the v3 target which includes `frsky_telemetry`. If not, you would have to upgrade the bootloader which is a more complex procedure.

rollysing, 2018-08-21:
Ok. I think someone needs to update the docs to let folks know it's not supported in Pixhawks that can only load v2 and/or 1MB limit.

Thanks for all the responses.

rollysing, 2018-08-21:
@bkueng During the firmware upgrade, it shows this. I'm presuming that means the bootloader is Version: 4? So it needs to be higher than that?
[image]

rollysing, 2018-08-21:
Just wanted to update before closing this. Phillip with mRobotics went above and beyond product customer service and updated the bootloader of the mRo Pixhawk to version 5. This now allowed QGC to recognize and upgrade using the Developer version of px4fmu-v3_default.px4. I can confirm that the Q X7 was able to discover all available sensors and data transmitted in real time.

@bkueng Thank you very much for all the guidance and directions.

closed 2018-08-21 by rollysing

bkueng, 2018-08-22:
Great! Good to see it works now.

hamishwillee, 2018-08-27:
> Ok. I think someone needs to update the docs to let folks know it's not supported in Pixhawks that can only load v2 and/or 1MB limit.

It's a reasonable request, but given that such a lot disabled in fmu_v2, hard to to. For many drivers we do have a comment that users should check that the module is present in firmware.

bkueng, 2018-08-27:
This is something that you have to change in the Taranis, or the display script that you use. It is not part of the PX4 Firmware.

fx3821, 2018-09-14:
hi，everyone
I have had similar problems recently and I'm glad to see this post and comments.
But i still have a question,
X12s was able to discover all available sensors and data transmitted in real time.
but the YAAPU LUA scripts did not works,
I would like to ask if any of the supported scripts are recommended?
or what should i do?
Thanks a lot

Here are my setup:
Frsky horus X12S
Pixhawk4 with PX4 1.8.0
X8Rreceiver

bkueng, 2018-09-14:
Hi
Someone would have add support for PX4 to the YAAPU script, as it is specifically designed for ArduPilot.
I know that LuaPilot works with PX4 and you find instructions here: https://docs.px4.io/en/peripherals/frsky_telemetry.html#taranis---luapilot-setup.

zhaoxiaowei1013, 2019-06-28:
> 嗨，
> 我最近遇到过类似的问题，我很高兴看到这篇文章和评论。
> 但我还有一个问题，
> X12s能够发现所有可用的传感器和实时传输的数据。
> 但是YAAPU LUA脚本不起作用，
> 我想问一下是否建议使用任何支持的脚本？
> 或者我该怎么办？
> 非常感谢
> 
> 这是我的设置：
> Frsky horus X12S 
> Pixhawk4与PX4 1.8.0 
> X8Rreceiver

px4 don't support yaapu lua scripts。

rollysing, 2019-06-28:
@zhaoxiaowei1013
> px4 don't support yaapu lua scripts。
It can but you need to use a [Teensy 3.2](https://www.pjrc.com/teensy/teensy31.html) with [Eric Stockenstrom’s](https://github.com/zs6buj/MavlinkToPassthru) firmware. Read the [developments](https://www.rcgroups.com/forums/showthread.php?3020527-An-Ardupilot-frsky-telemetry-LUA-script-for-the-Horus-X10-X12-and-Taranis-X9D-E-QX7). I have a Jumper T16 (works like Horus X10/X12) and it works very well.

Good luck.

zhaoxiaowei1013, 2019-06-29:
> @zhaoxiaowei1013
>
>
> Good luck.
I am devloping  PX4 firmware to support yaapu lua scripts ，and  thank your suggest。

rollysing, 2019-06-29:
@zhaoxiaowei1013 Is it your intention to write the code so it connect directly to the SPort of FrSky Rx without needing a Teensy or [Telemetry Cable ](http://www.craftandtheoryllc.com/store/ct-telemetry-cable-for-frsky-radios/) and natively work?