PX4/PX4-Autopilot #9738: no Pixhawk4 GCS connection after 1.8 upgrade

billzyoom, 2018-06-21:
I received my Holybro pixhawk4 two days ago and have had it up and running with stock 1.73 code. Flashed it today with qGroundControl running macOS over usb and have not been able read parameters since. My pc and Macs see the fc when using ap planner, qgc, mission planner and can even reapply stable firmware successfully using qGroundControl. It flashes successfully but then no response on the ph4. I've tried powering with power control board and usb. When rebooted, the ph4 is left with the following lights on...

I/O
   PWR - solid green
   B/E - solid amber
   ACT - flashes blue 

FMU - All lights off

[image]

qGroundCotrol offers ArduPilot but can't list any available options stable or otherwise.
Removed SD card from fc and tried rebooting.
Tried installing beta releases but non listed

How does one find and download holybro's factory image? I feel like I'm missing something...

[link](https://youtu.be/sBEi8i6831Y) to video here.

LorenzMeier, 2018-06-21:
Just confirming: you have flashed PX4? Try flashing again.

billzyoom, 2018-06-21:
Thanks for replying. I am flashing PX4 as offered by qGC. It also offers ArduPilot but does not list any available. I've flashed it over 10 times trying various ideas. It only successfully flashes the PX 1.8.

[image]

billzyoom, 2018-06-22:
I tried flashing my Holybro PH4 again today with qgroundcontrol 3.2 with same results. I then downloaded qgc 3.3.2 from daily builds which revealed the 1.7.4 firmware which wasn't there before. I was able to flash the 1.7.4 beta but have same results.

After flashing the I/O Power light stays green, B/E light blinks amber, ACT light blinks blue and FMU PWR light turns on, blinks very fast for 3 seconds and then turns off with some rapid beeps and then appears to reboot over and over and eventually stops rebooting with B/E blinking and ACT blinking and FMU PWR solid green. 

I'm attaching the app logs from qgroundcontrol 3.3.2.

[ph4_brick.txt](https://github.com/PX4/Firmware/files/2128505/ph4_brick.txt)

billzyoom, 2018-06-29:
I've spent over 8 hours troubleshooting this Pixhawk4 and I think it is bricked. I've emailed holybro product support email (prodectservice@holybro.com) 4 times over the last few weeks with no response. I purchased a PixHawk 2.1 Cube from ProfiCNC to keep the project moving forward which is working great. 

I've heard good things about Holybro and am looking for suggestions how I can get their attention. I don't want to resort to opening a claim with PayPal to get my money returned. 

All I can say is that it was working for about 30 minutes until I applied the 1.8 firmware at which point I can no longer communicate with the device. 

*sigh*

LorenzMeier, 2018-06-29:
Looking at the led pattern and the fact that you’ve successfully flashed v1.8 and it’s not communicating I would ask for a RMA. @jamming Could you help here with your customer?

I would love to have the board back for a post-mortem analysis.

billzyoom, 2018-09-19:
Holybro sent a new board but didn't ask for the old board back. If you provide a shipping address, I will send it over.

mhkabir, 2018-09-19:
FWIW, I had this same issue, and it turns out that my board had hardfaulted (due to driver development I was doing), and then it got "stuck" waiting for keyboard input on the console to clear the hardfault. 

The fault wouldn't/didn't clear itself over multiple reboots, leading to a "bricked" board. I needed to solder on a console and manually clear the fault.

FYI @dagar @davids5

dagar, 2018-10-24:
@davids5 could we somehow have a fresh flash clear the hardfault?

davids5, 2018-10-24:
@billzyoom - A quick test to see if the issue you had was the same as @mhkabir is to boot only connected to USB. If the PWR led does not flutter (CPU is running bootloader) then the board is damaged and not what @mhkabir experienced. What he noted is fixed in #10756

davids5, 2018-10-24:
@billzyoom - I would be happy to troubleshoot the issue if you wish to send me the board. Jut contact me as david_s5 on slack or skype.

closed 2018-10-25 by LorenzMeier