PX4/PX4-Autopilot #9739: Oneshot PWM_RATE = 0 with Firmware 1.8.0

Ralfde, 2018-06-22:
When I use the Pixracer board with 1.8.0 stable firmware and adjust PWM_RATE to 0 what should be Oneshot, then only the rate mode works and no yaw function is working.
When I go back to PWM_RATE 400Hz, everything works normal again.

250 size drone

LorenzMeier, 2018-06-22:
@bkueng Could you cross test?

mhkabir, 2018-06-22:
I'm running OneShot on a racer, flashed fresh with v1.8. Works fine for me.

Ralfde, 2018-06-22:
First I was flying with 400Hz, everything ok.
Then PWM_RATE = 0. No yaw and only rate mode.
Then I went back to PWM_RATE = 400 Hz, and everything was okay again.

mhkabir, 2018-06-22:
What do you mean by "no yaw and only rate mode"?

Please upload logs comparing before / after.

Ralfde, 2018-06-23:
All my programmed flight modes did not work and there was no yaw function on the stick function. Only roll and nick did work.
I will check, if I can find the logfiles.

Ralfde, 2018-06-23:
Found the log files, but I don't know which one was which flight. I hope you can see this.
[2018-06-22.zip](https://github.com/PX4/Firmware/files/2129723/2018-06-22.zip)

taileron, 2018-09-06:
... and OneShot (PWM_RATE = 0) lowers the PWM frequency from stable 400hz to very unstable 250hz instead of increasing it to something meaningfully fitting between 1 KHz and 4 KHz so PX4 is not really taking benefit of OneShots 10 times narrower pwm rate ?

mhkabir, 2018-09-06:
@davids5

davids5, 2018-09-08:
@taileron Was the the fmu started as a task? 

I would expect the update rate is on change. @dagar What is the mixer output rate?

taileron, 2018-09-09:
V1.80 FMU as task on a PixRacer 400Hz standard pwm currently works the best. I only wanted to give OneShot a try and measured the pwm outs with an oscilloscope out of curiosity ...

closed 2019-02-04 by stale[bot]