PX4/PX4-Autopilot #9741: Smooth transition between goto commands from map

feemi, 2018-06-22:
If we give go to command from fly map to opposite direction while vehicle is moving with normal speed. It tries to brake immediately which cause unstable flight.

Is it possible to wait for a bit slow down and have a smooth transition to reposition command ?

**Steps To Reproduce**
Send Vehicle to one direction.
Tap a point from map which is exaclty opposite direction while vehicle moving.
Send Go to command to new point.

----
**Enviroments**
- Daily QGC or stable version
- Autopilot board: Pixhawk 2
- Autopilot firmware: 1.7.3 or 1.8.0

LorenzMeier, 2018-06-23:
@Stifael are the tuning parameters fully documented by now?

LorenzMeier, 2018-06-23:
In general if the flight becomes unstable then your vehicle is not set up correctly and you have a more fundamental problem. It should just be very knife-edge, but not unstable.

Stifael, 2018-06-25:
The tuning parameter for auto are summarized [here](https://docs.px4.io/en/advanced_config/advanced_mc_position_tuning.html).

About instability: 
>  if the flight becomes unstable then your vehicle is not set up correctly and you have a more fundamental problem.

Agreed. @feemi, you should follow the [tuning guide](https://docs.px4.io/en/advanced_config/pid_tuning_guide_multicopter.html) for multi-copters.

About `Go-To` command:
Currently the implementation is such as you have described with instantaneous direction change. For mission, however, it will behave as you have requested: the vehicle will slow-down close to first waypoint to guarantee a smooth trajectory. In principle this could also be accomplished with a `Go-to` commands, but I am not convinced that this would be the right thing to do mainly due to safety in situation where an instantaneous flight direction change is required.

feemi, 2018-06-25:
I think description of my problem is little bit wrong. The unstable flight that i am try to describe is about size of our drone and movement limitations. In mission mode or in position mode vehicle moves very smooth with current limitations. But go to command is knife edge and it is too much for our structure. Probably there is no such issue with small drones since they are very agile. I am just wondering would that be useful if there would be prior slow down option before knife edge brake when vehicle gets go to command?

BazookaJoe1900, 2019-01-03:
@feemi @LorenzMeier @dagar 
I see the same issue in my vehicle, and its not nice at all seems like it going to flip on every turn.
I understood it fixed on version 1.9, but for version 1.8 it seems critical.
is there a possibility to 'cherry pick' the fix into new version 1.8.3?
or, is there a workaround that can mitigate the issue by setting some parameters?

BazookaJoe1900, 2019-01-27:
@feemi @LorenzMeier @dagar
can you please comment?
what can I do until 1.9 will release?

BazookaJoe1900, 2019-02-03:
@feemi thanks! I will try that.
I reduced the roll_rate to 120deg/sec before, I will reduce it more

BazookaJoe1900, 2019-02-03:
@feemi, there are many changes between 1.8 to master (1.9 TBD) and I am not sure which issue solved the harsh pitch/roll, but in the end it doesn't exist on 1.9, you can see that in the simulator.
BTW, why do you think that the PID need to be retune, I think that that rate_max's are just limiters, and should not affect the PID unless the drone really need this "high" rates from time to time.

BazookaJoe1900, 2019-02-12:
@feemi I think the issue still exist in v1.9 beta, if you look at [this log](https://logs.px4.io/plot_app?log=7ebec132-efe6-49ee-91c9-336cfe1ebde9) you an see this hard roll when turnning.

do you think its also max roll issue there?

[image]

bresch, 2019-07-10:
@BazookaJoe1900 The roll error you see in that plot is an artefact due to the Euler attitude representation. For example: the shortest way to move from a 45° pitch, 0° yaw to -45° pitch, 180° yaw is to perform a pure rotation around the vehicle's Z axis, no pitch change is required. This is why you see large pitch/roll tracking errors during sudden yaw changes.

For the rest, the aggressive change of speed is fixed in 1.9 with the new jerk-limited auto mode.

closed 2019-07-10 by bresch