PX4/PX4-Autopilot #9746: After uploading a "bad" mission, Bad tones (beeps) happen everytime, even when uploading a good mission

Antiheavy, 2018-06-22:
When the vehicle rejects a mission the first time we hear the "bad" tones from the autopilot beeper (this is expected).  Any subsequent uploads of ANY mission (good or bad) the "bad" tones happen.

This is happening on 1.8.0.

I think has been happening for a long time, but we finally just realized it is 100% repeatable.  The way we usually see it is we upload a mission with a landing pattern that is too short and it gets rejected.

dagar, 2018-06-22:
Something must be wrong with the logic here. 

https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2069-L2075

Possibly a previous warning that hasn't been cleared.

Antiheavy, 2018-07-10:
@dagar the tweak we just tried seems to have resolved this issue.  Is it worth making a smaller stand-alone PR here in upstream?

Antiheavy, 2018-08-13:
@dagar  Did the fix for this make it upstream?

dagar, 2018-08-13:
No, the more thorough fix is here. https://github.com/PX4/Firmware/pull/9747

Antiheavy, 2019-01-05:
@dagar the fix for this bug seems like something that should be part of v1.9.0.   thoughts?

Antiheavy, 2019-06-25:
still valid issue?

dagar, 2019-06-25:
This isn't fixed in master?

Antiheavy, 2020-01-09:
Yep, this is fixed in Master.  closing.

closed 2020-01-09 by Antiheavy