PX4/PX4-Autopilot #9750: How to enable odometry using EKF2 estimator in v1.8.0 ?

gajena, 2018-06-23:
I am sending odometry on `/mavros/odometry/odom` and not getting any changes in local_pose.

gajena, 2018-06-23:
I am having one more problem when I am using EKF2 aid_mask as vision position suddenly estimator getting stop.
Should I use this branch of ecl lib `pr-extVisInit`?

mhkabir, 2018-06-23:
Yes.

gajena, 2018-06-23:
Now estimator is initializing but when I started to publish on `vision_pose` it's coming on `local_pose`
but after sometime `local_pose` x and y becoming zero and z become constant at last value.

mhkabir, 2018-06-23:
The estimator is probably rejecting your data after some time. Please open a new issue with logs and details about your vision system.

RomanBapst, 2018-06-24:
@gajena You should not be using any branches in ecl. It should just work with the release or latest master. As Kabir said, a log file would enable us to help you.

mhkabir, 2018-06-24:
@RomanBapst Vision fusion works fine on ecl master/release, however the branch I referred to improves initialization and fixes the case where the visual estimator resets in flight. See https://github.com/PX4/ecl/pull/464.

gajena, 2018-06-25:
@RomanBapst @mhkabir [Logs]( https://review.px4.io/plot_app?log=5d0f3bbc-a6a4-4c43-901d-ceadff5999d2#Nav-Local-Position-X )
I am done this test using ROVIO and I am not moving quad just moving VIO sensor.

mhkabir, 2018-06-25:
> Yes, of course. The EK

mhkabir, 2018-06-25:
Yes, they need to be a rigid body and be moved together, otherwise the EKF will reject data.

gajena, 2018-06-25:
Okay, I will test this.
And to give commanded position should I use this topic in mavros `mavros/setpoint_position/local` ?

tiralonghipol, 2018-08-23:
@gajena  HIi ! Any update on this ?
Thank you