PX4/PX4-Autopilot #9751: Unable to write a custom C code for tune library

Ankur1014, 2018-06-24:
**Describe the bug**
Till now I was using FW 1.6.5. I had my own custom C code for tunes that was basically producing a custom tune at startup for Pixhawk 1. My C file included the <drivers/drv_tone_alarm.h> file. I upgraded to FW 1.8.0. Now the header file <drivers/drv_tone_alarm.h> includes another header file <lib/tunes/tune_definition.h> that has defined the enum as a class. Thus it is causing compilation issue in my C code. I am not very proficient in C++ code and do not want to switch to C++. Also I do not see the need of defining the tone as enum class. An enum should suffice.

**To Reproduce**
Write a C code and include <drivers/drv_tone_alarm.h> . Build the C code . The compilation would fail

**Expected behavior**
Users should be able to include the headers in both C and C++ and firmware should build

Ankur1014, 2018-06-25:
Hi LorenzMeier,
Thanks for responding. So here is the issue. I have a custom module which I have added to my rc.txt so that mdoule starts up by default at power up. The module basically executes a sequence of LED blinking and some Tune sequence upon boot up. I had coded the module in .c

Till now I was including the header <drivers/drv_tone_alarm.h> and using the enums defined in that header.
Upon upgrading to 1.8.0 , I found that this header is now including one more header <lib/tunes/tune_definition.h> . In this new header another set of enum is developed but this is declared as 

`enum class Tune {
...
}`

I do not mind the new set of enum and I am happy to use them. The problem is that they have been defined as `class` which is causing compilation issue in my C code. I do not know the reason why this has been coded as `enum class` and why not just `enum`. Even it is necessary to do that, a `ifdef _cplusplus` guard can be added to support C coding.

Let me know if that helps.

bkueng, 2018-06-26:
Hi

As most of our code is C++, I suggest you just rename your module to .cpp.
We use an `enum class` because the enum names are in a local namespace leading to less naming conflicts and improved readability.

closed 2019-02-04 by stale[bot]