PX4/PX4-Autopilot #9752: FW landing: flare curve documentation outdated / misleading

tstastny, 2018-06-24:
Disclaimer: Maybe this should be an issue on devguide -- but it also pertains to the comments in the firmware, so I put it here, please let me know if I should move it.

Issue:

1) The FW_LND_HVIRT param is described in [.../fixed_wing_landing.html](https://docs.px4.io/en/flying/fixed_wing_landing.html) as "[the] Virtual horizontal line/altitude used to calculate the flare trajectory. This represents the sub-ground altitude where the vehicle would reach the desired post-flare touchdown position, if it were to maintain its initial descent trajectory." Looking at the figure there, this text seems to imply (or at least can be interpreted) that "maintaining its initial descent trajectory" refers to the glide slope before the flare. This glide slope intersects the touchdown x-position and FW_LAND_HVIRT y-position (negative, under ground) in the figure as well. However, if one actually plots this (see below -- note, this was plotted using the firmware equations and default param values), you can see this is actually not the case. More likely the param is referring to the line underground which the exponential curve asymptotically approaches. But in any case, the figure is clearly drawn wrong - the flare curve does not smoothly approach the touchdown point as shown in the docs, and the glide slope does not intersect the virtual line at the touchdown x-position, but will be shifted some amount depending on the flare params.

[image]

2) The source code reference to this (potentially wrong) figure in the docs also is outdated, see [.../Landingslope.hpp#L48](https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/Landingslope.hpp#L48). And, there is no param description given in the params c file [.../fw_pos_control_l1_params.c#L259](https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_params.c#L259).

If the figure shown in the docs is actually what is intended, then the flare curve calculation needs to be reworked. Please let me know if I'm missing something -- otherwise I could make a PR.

LorenzMeier, 2018-06-24:
@thomasgubler this looks like it is a question for you

hamishwillee, 2018-06-24:
FYI I created that diagram based on a much older diagram and discussion with various developers - not the code. It is possible that everyone's understanding is based on the older image rather than actual behaviour as inferred from code. Interest to see what @thomasgubler says.

> and the glide slope does not intersect the virtual line at the touchdown x-position, but will be shifted some amount depending on the flare params.

Fair enough. Perhaps this should be represented as some sort of spread of possible landing paths? Does whatever tool you are using to plot what the actual equations are doing have a front end we can place with to see how things vary?

When the correct behaviour is established, can we have a [corresponding defect](https://github.com/PX4/px4_user_guide/issues/new?title=Doc+Bug:+Landing%20(Fixed%20Wing)%20%C2%B7%20PX4%20User%20Guide&body=DESCRIBE+PROBLEM+WITH+DOCS+HERE%0A%0ABug+Page:+[Landing%20(Fixed%20Wing)%20%C2%B7%20PX4%20User%20Guide](https%3A%2F%2Fdocs.px4.io%2Fen%2Fflying%2Ffixed_wing_landing.html)) created in the user guide please.

thomasgubler, 2018-06-25:
Here is the original diagram that the comments in the code refer to:
[image]

tstastny, 2018-06-25:
> Here is the original diagram that the comments in the code refer to:

Ok - looks like this is intended behavior then. So @hamishwillee it would probably be good if we update this figure onto the documentation (either @thomasgubler's figure or I could generate another one like I posted but where the flare curve intersects properly with the straight line). Then the the firmware comments don't need to be changed, and only the dev guide description for hvirt could possibly be slightly modified. wdyt?

> Does whatever tool you are using to plot what the actual equations are doing have a front end we can play with to see how things vary with the flare params?

I was just plotting with matlab. I think the main thing for useability (probably a separate discussion) is understanding what the flare length turns out to be for a given flare height and virtual line. That is the part that is not necessarily so intuitive when tuning these params during flight -- different combinations of different flare heights and hvirt's are what give different flare lengths. (of course glide slope also plays into this). But you're right -- may be nice for users to have a tool to play with this and see what they are actually telling the plane to do before setting the params and trying it on the field. You could type the equations from the firmware into that link you provided. But I guess not everyone has matlab.

thomasgubler, 2018-06-25:
I imported the python notebook that I used to generate the plot back then:

https://drive.google.com/file/d/1j5Nq6aETXYSwZcw5qSXUGagkyNWRf_Nm/view?usp=sharing

It seems only half functional and some plot features are commented out but it might be a starting point

hamishwillee, 2018-06-26:
@tstastny @thomasgubler The reason I didn't use the original diagram was that it came without explanation
- it wasn't clear what any of the values are
- it wasn't clear how they mapped to parameters. 

The diagram is now in the user guide - it is intended primarily to enable users to understand what the landing slope looks like by default and how it can be tuned. 

So I don't think we should revert to the original diagram, because it does not answer those questions adequately. What I would like (pretty please) is if one of you could regenerate the image *for the default parameters* such that is accurate, includes the symbols, parameters (that are possible), nominal touch down point, and enough of the glide slope to see the bit before flare and that we're clearly heading for HVIRT. 
In the text we can then add info pointing to source for the symbol code. 

We would then create a separate issue to (eventually) add equations into the online plotter [Yes, I do not have matlab. I know that you can't rely on an online site having great longevity, but it will be useful for at least some time]

thomasgubler, 2018-06-26:
@hamishwillee that makes sense. The diagram was originally checked in the repo and was meant to help someone that reads the code to understand the equations. It is in that form definitely not suitable for a user because as pointed out the highlighted values do not correspond to the tuning knobs/parameters that are exposed to the user.

tstastny, 2018-06-26:
> because as pointed out the highlighted values do not correspond to the tuning knobs/parameters that are exposed to the user.

fyi - I'm working on a first cut of actually exposing these parameters on QGC. However, it's not exactly straight forward how to do this properly while also considering ardupilot and users who simply don't want to see 3 extra flare params on their screen etc. I'll link the branch once it is running.

thomasgubler, 2018-06-26:
Awesome. There are 2 things we can do:
1) exposing the params in the params tab so that a power use can change it. 
2) work with @dogmaphobic and @DonLakeFlyer to include a user friendly tab where a user can configure the landing approach (e.g. with sliders) without actually knowing about the params.

tstastny, 2018-06-26:
> work with @dogmaphobic and @DonLakeFlyer to include a user friendly tab where a user can configure the landing approach (e.g. with sliders) without actually knowing about the params.

See https://github.com/mavlink/qgroundcontrol/issues/6457#issuecomment-400324799 for WIP branch link. The rest of that thread has some elaboration on why this is challenging.

hamishwillee, 2018-06-27:
Presumably the main params are already exposed in the full parameter list (e.g. https://docs.px4.io/en/advanced_config/parameter_reference.html#FW_LND_FLALT ) - are you suggesting some other params be exposed? 
As you both say though, a dedicated user friendly tab for these settings will be great - something that shows visually the effect on the glide slope.

In the meantime, where does this leave us re the diagram? https://github.com/PX4/Firmware/issues/9752#issuecomment-400141832

tstastny, 2018-06-29:
> where does this leave us re the diagram and the text?

@hamishwillee what about this?

[image]

thomasgubler, 2018-06-29:
@tstastny Thanks a lot! :+1:  `FW_LND_ANG` landing slope angle is missing

tstastny, 2018-06-29:
Not sure how else to show the landing slope *angle*. wdyt?

[image]

tstastny, 2018-06-29:
[image]

ok corrected

thomasgubler, 2018-06-29:
In my diagram I just put it between the blue line and ground. To me that is more natural. And it would be `FW_LND_ANG` not `FW_LND_SLOPE`? Or are you planning to change that?

tstastny, 2018-06-29:
> In my diagram I just put it between the blue line and ground

ok

> Or are you planning to change that?

no, just a typo

[image]

hamishwillee, 2018-07-02:
Thanks very much. I have added the updated image with some minor text tweaks here: https://github.com/PX4/px4_user_guide/pull/288

Can you please confirm you're both OK with the changes?

hamishwillee, 2018-07-02:
PS Not something I would "fix" but the scale here does make it look like the FW powers towards the ground at a near 45 degree angle! :-)