PX4/PX4-Autopilot #9754: jenkins: build failing on OSX

lamping7, 2018-06-24:
Examples:
- http://ci.px4.io:8080/blue/rest/organizations/jenkins/pipelines/Firmware/branches/master/runs/283/log/?start=0
- http://ci.px4.io:8080/blue/rest/organizations/jenkins/pipelines/Firmware/branches/master/runs/285/log/?start=0

git checkout is step is having problems. The problem is localized to one slave - gus-mac.

To me, it looks like the local copy of the git repo is corrupt. The same specific errors happen across the multiple runs of the job on the same branch (master in the case above). The same workspace is used for all runs of the branch-job combo.

lamping7, 2018-06-24:
Fixed for now.
http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/master/286/pipeline/66

What I did was wipe the workspace by setting the property "Wipe out repository & force clone" in the configuration for the job in Jenkins. This cleared the corrupt repo off the slave allowing it to get a fresh copy. I'm not sure if we want to do this all the time, so I un-set it after this ran.

Here is where it is set:
[image]

We should discuss if we want this all the time (may hurt performance) or just clean the workspace manually when this happens as I just did. Thoughts @dagar ?

dagar, 2018-06-24:
I don't fully understand the implications between different builds and different slaves.

Does this apply to every single build? For example an existing PR that has another small commit pushed to it shortly after it's already built? These are the situations where the overall build is nearly twice as fast because it's mostly already cloned and the ccache hit rate is quite high.

dagar, 2018-06-24:
@dogmaphobic any ideas why the Jenkins PX4 clone might be corrupting on your mac?

lamping7, 2018-06-26:
Additionally, I just noticed that QGC build has been broken as well. Same OSX slave (gus-mac). The error is slightly different:
```
...
Resolving deltas:  73% (93/127), completed with 41 local objects.
[OSX Debug] fatal: pack has 34 unresolved deltas
[OSX Debug] fatal: index-pack failed
```
I did the same as above for the Firmware job to clean the workspace.

lamping7, 2018-06-29:
I think this may be what is causing a corrupt git repo on gus-mac (but I don't know what is causing this). It's terminating in the middle of a checkout step.
http://ci.px4.io:8080/job/Firmware/view/change-requests/job/PR-9802/2/consoleFull

```
java.io.EOFException
	at java.io.ObjectInputStream$PeekInputStream.readFully(ObjectInputStream.java:2675)
	at java.io.ObjectInputStream$BlockDataInputStream.readShort(ObjectInputStream.java:3150)
	at java.io.ObjectInputStream.readStreamHeader(ObjectInputStream.java:859)
	at java.io.ObjectInputStream.<init>(ObjectInputStream.java:355)
	at hudson.remoting.ObjectInputStreamEx.<init>(ObjectInputStreamEx.java:48)
	at hudson.remoting.AbstractSynchronousByteArrayCommandTransport.read(AbstractSynchronousByteArrayCommandTransport.java:36)
	at hudson.remoting.SynchronousCommandTransport$ReaderThread.run(SynchronousCommandTransport.java:63)
Caused: java.io.IOException: Unexpected termination of the channel
	at hudson.remoting.SynchronousCommandTransport$ReaderThread.run(SynchronousCommandTransport.java:77)
Also:   hudson.remoting.Channel$CallSiteStackTrace: Remote call to gus-mac
		at hudson.remoting.Channel.attachCallSiteStackTrace(Channel.java:1741)
		at hudson.remoting.Request.call(Request.java:202)
		at hudson.remoting.Channel.call(Channel.java:954)
		at hudson.FilePath.act(FilePath.java:1036)
		at hudson.FilePath.act(FilePath.java:1025)
		at org.jenkinsci.plugins.gitclient.Git.getClient(Git.java:137)
		at hudson.plugins.git.GitSCM.createClient(GitSCM.java:821)
		at hudson.plugins.git.GitSCM.createClient(GitSCM.java:812)
		at hudson.plugins.git.GitSCM.checkout(GitSCM.java:1180)
		at org.jenkinsci.plugins.workflow.steps.scm.SCMStep.checkout(SCMStep.java:113)
		at org.jenkinsci.plugins.workflow.steps.scm.SCMStep$StepExecutionImpl.run(SCMStep.java:85)
		at org.jenkinsci.plugins.workflow.steps.scm.SCMStep$StepExecutionImpl.run(SCMStep.java:75)
		at org.jenkinsci.plugins.workflow.steps.AbstractSynchronousNonBlockingStepExecution$1$1.call(AbstractSynchronousNonBlockingStepExecution.java:47)
		at hudson.security.ACL.impersonate(ACL.java:290)
		at org.jenkinsci.plugins.workflow.steps.AbstractSynchronousNonBlockingStepExecution$1.run(AbstractSynchronousNonBlockingStepExecution.java:44)
		at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
		at java.util.concurrent.FutureTask.run(FutureTask.java:266)
		at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
		at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
		at java.lang.Thread.run(Thread.java:748)
Caused: hudson.remoting.RequestAbortedException
	at hudson.remoting.Request.abort(Request.java:340)
	at hudson.remoting.Channel.terminate(Channel.java:1038)
	at hudson.remoting.SynchronousCommandTransport$ReaderThread.run(SynchronousCommandTransport.java:96)
Finished: FAILURE
```

lamping7, 2018-08-09:
Cause was not determined, but a cleanup of the local machine removed the issue for now.

closed 2018-08-09 by lamping7