PX4/PX4-Autopilot #9755: Gazebo fixed wing SITL nevers completes landing

DonLakeFlyer, 2018-06-24:
Using master firmware, master QGC:
* Create a simple mission with a few waypoints and a landing pattern
* Run it
* When the vehicle comes in to flare it slows way down and the altitude drops to 0 as it gets to the landing point
* The vehicle then speeds back up to full speed and continues past the landing point flying at 0 alt

This is a big problem for me since it means I can't test resume mission in QGC.

LorenzMeier, 2018-06-24:
@dagar Could you have a look?

dagar, 2018-06-24:
I believe this is a result of the limited or lack of ground friction, overpowered sitl plane model, and throttle settings.

I've been meaning to get a full plane auto mission (runway takeoff, waypoints, loiter, landing) in the CI system.

lamping7, 2018-08-18:
Does https://github.com/PX4/Firmware/pull/10250 fix this for you @DonLakeFlyer? It's merged in master.

closed 2019-02-04 by stale[bot]