PX4/PX4-Autopilot #9757: Add LPE back to CI

jgoppert, 2018-06-24:
LPE used to run on CI with  a ROS mission flight in SITL  but after the new jenkins changes I think it got wiped out. Due to the current user base, we should add this back if possible. @dagar

TSC21, 2018-06-24:
Addressed here a4a4192.

jgoppert, 2018-06-24:
We should cherry-pick this and create a new PR to make sure master still works.

dagar, 2018-06-24:
@TSC21 failing http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/PR-9301/33/pipeline

That's fine for now, but we need to be realistic about support. The other thing that concerns me is that the estimator is becoming more tightly integrated with commander for various checks, failsafes, etc. EKF2 and LPE can't even agree on estimator_status usage (they each use different fields), so I don't hold out much hope for LPE being a good or even safe user experience without additional effort.

TSC21, 2018-06-24:
> @TSC21 failing http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/PR-9301/33/pipeline

Yes I know. Current on lab experiments with it.

> The other thing that concerns me is that the estimator is becoming more tightly integrated with commander for various checks, failsafes, etc.

I don't really agree with this. PX4, as an open-source effort which allows a good experience platform for companies developing their products but also on the academic world, should be too much specific or too much coupled for a single specific estimator or control environment, but rather be as agnostic as possible, in order that someone that wants to couple their own esitmator or control routine is able to do it without having to redo the entire pipeline. An example of that is having specific uORB msgs as `ekf2_innovations`. Why not just `innovations`? What's the reason to taper into a specific framework and not having instead a more generalized, agnostic and safe framework where we one can plug their own apps? I think, since the begining, that was one of the main reasons of PX4 to exist.
Being realistic about support is actually also no when to deprecate things, and that's also one of the things I will be working with @jgoppert in the next few weeks. LPE should continue to be supported in the mean time, as there is a big community still using (that I know of), but soon we should start deprecating it.

dagar, 2018-06-24:
@TSC21 I agree with you, I'm simply stating the current state. The estimator checks both preflight and during have become extremely valuable and this will only get more sophisticated. For this to be agnostic we need people to design and agree on interfaces.

The important thing would be agreeing on the interface that commander is using for all of these checks (currently estimator_status). It's not okay for it to contain completely different fields for each estimator to present the same information like what we have now for sensor health.

I'd be happy to ensure commander and even things like flight review respect the agreed upon interface, but I can't worry about every single estimator conforming.

jgoppert, 2018-06-24:
It is pretty hard to maintain an estimator agnostic API when we only test one estimator in CI, so I think we should maintain at least 2, even if the other estimator is not officially supported and even a toy example.

One could argue about the benefit of having an estimator agnostic API, but I think we are all on the same page. Having the agnostic API allows continuous development and is critical for appeal to the research community.

dagar, 2018-06-25:
The CI tests are too basic right now to catch anything but the most blatant failures. Calling it an agnostic API might have been over stating it. The published estimator status doesn't need to cover everything, it just has to stop being used completely differently. It could also contain flags for which parts are valid.

We can start an issue listing some of the worst offenders if you're interested. I wouldn't be opposed to splitting the existing message into lpe_status + ekf2_status and starting again.

jgoppert, 2018-06-25:
The ros test is pretty decent actually.  It has a comparison to the ground
truth after it flies a mission and fails if the error is too high.

I don't really want to add another message for lpe. We will be releasing a
series of new estimators soon and that is why we want to slowly deprecate
LPE. These estimators use a common api to compare several algorithms for
research purposes. If you can work towards documenting and stabilizing the
ekf2 api that should be enough. If we had a separate message it would
actually be harder to maintain think. The message is a good insulation
point. Adding some extra valid flags may be a good idea. We are already
flying using the existing messages for the new estimators.

If you can list what flags are incorrect in LPE that would be helpful. It
would also help me ensure I catch any repeated errors in the new estimators

dagar, 2018-06-25:
> The ros test is pretty decent actually. 

Not really, it's only testing perfect setups in good conditions. We need testing that exercises a wide range of annoying real world problems and failures at all stages.

I was only talking about adding a message for LPE as an incremental step towards an estimator agnostic message. That's why I suggested doing it for EKF2 as well, then only readding pieces with appropriate usage system wide or flags so that any consumer of estimator_status can use it properly. There are issues I have with the EKF2 side of it as well.

> If you can list what flags are incorrect in LPE that would be helpful. It
> would also help me ensure I catch any repeated errors in the new estimators

The flags used by commander and preflight check are what I was originally referring to for tighter integration and a safer system, but LPE missing them isn't necessarily wrong.

As a start I'd take a look at LPE's usage of estimator_status health_flags and timeout_flags. Finding a unified solution would be a good first step to being able to offer minimal support. I'd actually like to get these out of bitfields entirely, but that's another story. As we continue to lock down commander and extend flight review these are only going to grow.
 
Examples of inconsistencies
 - https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1755-L1822
 - https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L2007-L2028
 - https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L2089-L2110
 - https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp#L540

LorenzMeier, 2018-06-25:
The main issue is not the interface, but getting enough usage in LPE and an active maintenance community around it. Given that we would be having different discussions. Maintaining a second estimator is totally worth it, but not just for the sake of generality of the interface.

dagar, 2018-06-25:
Another issue that just occurred to me is the sensor_bias message being published only by EKF2. Several modules (including the land detector) are using accel_{x, y, z} directly from this message.

TSC21, 2018-06-25:
@dagar can you please point out what is lacking on LPE currently for fully integration with the PX4 frameworks and pipelines? As you may imagine, there was some stuff that was left behind given that we were not able to update the interface and maintain it. Though I will glady add what's missing and fix whatever need. If you could open an issue for that I would then open a PR to fix it.

dagar, 2018-06-25:
I don't fully know yet, but I'd start with my last comment and go from there. We can discuss some of the higher level points on the dev call.

jgoppert, 2018-06-25:
@LorenzMeier I think it is just time to deprecate LPE like we did INAV. I am more interested in working on new algorithms and EKF2 is now providing a viable indoor navigation routine. Dropping LPE will free me to focus on the more interesting work. I think we see a new estimation algorithm every couple of years, and we need to work on a smooth deprecation process. For my research all I want is a stable estimation message and some good documentation on how to us it.

closed 2018-06-26 by dagar