PX4/PX4-Autopilot #9761: EKF2 GPS Fusion Drifts

cimbar, 2018-06-25:
I was testing EKF2 in current master for OFFBOARD flight. The external commander gives same path for each circumstances (different EKF_AID_MASKs). GPS only and Vision only flights work properly in SITL but when GPS is fused with IMU or Vision position; GPS drifts and mismatch occurs between local position and vision estimate. Therefore GPS fused flights can not complete OFFBOARD mission. Is it related with the EKF parameter tuning or a possible bug? I was using same environment in 1.7.3 version with LPE and there wasn't any problem.

**EKF_AID_MASK: 1 GPS** - Test Result: PASSED
[image]
https://logs.px4.io/plot_app?log=b950b0a1-2d07-43c3-be11-aa1dd93e9868

**EKF_AID_MASK: 5 GPS+IMU** - Test Result: FAILED
[image]
log: https://logs.px4.io/plot_app?log=febcceab-3ad9-4a33-a541-cde225d59afc

**EKF_AID_MASK: 9 GPS+Vision Position** - Test Result: FAILED
[image]
log: https://logs.px4.io/plot_app?log=02f87050-8597-4d33-af98-5b0b8ec0c184

**EKF_AID_MASK: 24 Vision Position+Vision Yaw** - Test Result: PASSED
[image]
log: https://logs.px4.io/plot_app?log=5186f402-c602-4dde-a8b5-4bc243c41cda

**EKF_AID_MASK: 25 GPS+Vision Position+Vision Yaw** - Test Result: FAILED
[image]
log: https://logs.px4.io/plot_app?log=534bf199-e201-4d3a-b7d1-625887cc885e

**EKF_AID_MASK: 28 IMU+Vision Position+Vision Yaw** - Test Result: PASSED
[image]
log: https://logs.px4.io/plot_app?log=5855ca61-0fe7-4d6a-a746-eb0ff1b73d58

**EKF_AID_MASK: 29 GPS+IMU+Vision Position+Vision Yaw** - Test Result: FAILED
[image]
log: https://logs.px4.io/plot_app?log=c260fa4e-b727-41cb-84af-e32a488f5afc

tiralonghipol, 2018-09-27:
Hi. Can you describe more how do you provide the vision_position and vision_yaw fusion?
I am having truble with the EKF2 too, maybe you can help me
Are you sentingo pose_stamped msgs using mavros over /local_position/pose ?

cimbar, 2018-10-01:
> Hi. Can you describe more how do you provide the vision_position and vision_yaw fusion?
> I am having truble with the EKF2 too, maybe you can help me
> Are you sentingo pose_stamped msgs using mavros over /local_position/pose ?

I am using /tf listener but If you wish to send pose stamped message, you should publish it to /mavros/vision_pose/pose topic. /local_position/pose is a topic that publishing from mavros so it is the position information of the vehicle. You can check mavros_extras  [wiki page](http://wiki.ros.org/mavros_extras#vision_pose_estimate) to understand which topic is used as whether a publisher or subscriber.

tiralonghipol, 2018-10-01:
yes I am using exactly the same 'configuration' as you described, 
but can you be more specific ? like: 
1) the feeding frequency 
2) what parameters did you change (if any) in qgc

thanks again

cimbar, 2018-10-01:
You should activate necessary bitmasks in EKF2_AID_MASK as vision position and vision yaw. What kind of problem do you have with vision position? Can you see any vision_position data in QGC Mavlink Inspector?

tiralonghipol, 2018-10-01:
ok it is quite complicated but I'll try to make it clear:

- If I publish on mavros/local_position/pose the PoseStamped which is received, it is good @5Hz
(hls_lfcd_lds_driver). I am using hector_slam and If the movement is too fast the EKF2 sometimes fail

- If I use the same configuration but with RPLidar a3 (rplidar_ros), after some times, tipically between 30 and 50 seconds, the ekf2 completely fails and the LOCAL_POSITION_NED.* goes to zero. 
Just for test, I tried both to fuse vision or not, and the behaviour is that when the filter fails the z stil good if I'm setting vision as the primary source of height.

Also, I tried with a camera, publishing something around 20Hz, and the behaviour is almost the same

Hope to be clear lol

closed 2019-02-04 by stale[bot]