PX4/PX4-Autopilot #9768: Descending with MPC_LAND_SPEED instead of MPC_Z_VEL_MAX_DN

ivodre, 2018-06-26:
**Describe the bug**
When descending during a mission in simulation the Deltaquad does the whole descend with MPC_LAND_SPEED instead of descending with MPC_Z_VEL_MAX_DN until MPC_LAND_ALT1, and than ramping down to MPC_LAND_SPEED. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create "multicopter" mission with delta quad in gazebo (e.g. at 30m altitude)
2. Make sure to set reasonable values for MPC_LAND_SPEED, MPC_Z_VEL_MAX_DN, MPC_LAND_ALT1, MPC_LAND_ALT2
2. Start mission

**Expected behavior**
The vehicle climbs with MPC_Z_VEL_MAX_UP, flies at desired altitude and than descends with MPC_LAND_SPEED all the way to the ground. 

**Log Files**
https://logs.px4.io/plot_app?log=70e15793-db80-4ec7-91d8-d1b6af522e70

**Drone (please complete the following information):**
- Gazebo Deltaquad

**Additional context**
We observed this also in real flights

@lamping7

dagar, 2018-06-26:
@sanderux @RomanBapst

dagar, 2018-06-26:
Likely an issue with the multicopter position controller rather than VTOL specific.

closed 2019-02-04 by stale[bot]