PX4/PX4-Autopilot #9775: Failsafe PWM not set

knopancakes, 2018-06-26:
The following line is erroneous:

pwm failsafe -d ${OUTPUT_DEV} ${FAILSAFE}

The PWM driver expects a -p before the failsafe value. Additionally, the failsafe command expects a -m or -c argument, which is not provided here.

mcsauder, 2018-07-17:
Thanks @knopancakes ,

Can you let me know if the following gives you the expected behavior at that line? 
`pwm failsafe -c 12345678 -p ${FAILSAFE} -d ${OUTPUT_DEV}`
or, 
`pwm failsafe -c ${PWM_OUT} -p ${FAILSAFE} -d ${OUTPUT_DEV}`

If either line does I can add it to my current startup script work.

Thanks.

-Mark

knopancakes, 2018-07-17:
I can confirm that providing the -p flag is the correct fix.

`pwm failsafe -c ${PWM_OUT} -p ${FAILSAFE} -d ${OUTPUT_DEV}`

knopancakes, 2018-07-17:
Unfortunately I do not have hardware with me at the moment so I cannot verify the script, but I am pretty sure

closed 2019-02-04 by stale[bot]