PX4/PX4-Autopilot #9776: PWM_RATE not set with -c argument in rc.interface

knopancakes, 2018-06-26:
If the user creates a config.txt under /fs/microsd/etc/, and that config.txt contains the following:

set PWM_OUT 12345678
set PWM_RATE 0

The following line in rc.interface silently fails:

https://github.com/PX4/Firmware/blob/0ac1ef12c77ef5a0db05d22bab4cfbd61b0c24a6/ROMFS/px4fmu_common/init.d/rc.interface#L85

The reason is due to the -c argument. Using instead -m 0xFF has the desired effect.

My workaround was to explicitly call the following in extras.txt:

```
set PWM_OUT 0xFF
set PWM_RATE 0
pwm rate -m ${PWM_OUT} -r ${PWM_RATE}
```

mcsauder, 2018-07-17:
Hi @knopancakes , 

Could you pull down the latest master branch and let us know if it continues to fail in the same manner?  I am working on the startup scripts right now and am interested in addressing this issue with you.

Thanks.

-Mark

knopancakes, 2018-07-17:
Unfortunately I do not have hardware with me at the moment so I cannot verify this behavior, but in retrospect it may be related to user error (i.e. my not understanding pwm group sets)

knopancakes, 2018-07-17:
I will get back to you on this when I have access to the hardware again

mcsauder, 2018-07-20:
Hi @knopancakes , I think you'll find your issue addressed in this PR: https://github.com/PX4/Firmware/pull/9936

Let us know if we can close out this issue or if anything remains unattended to!

-Mark

mcsauder, 2018-08-05:
@knopancakes , ping.

closed 2019-02-04 by stale[bot]