PX4/PX4-Autopilot #9777: TECS, Landing, flaps and minimum speeds

ryanjAA, 2018-06-27:
This is reproducible on any version.

FW_airspd_min (minimum speed) is not a variable parameter but when utilizing things like flaps, it actually should be since they lower the effective minimum speed that can be flown it. Since we do not have speed based flap deployment we are stuck with whatever the min speed set is. The problem arises because the TECS stall speed code (which is great) won't let you fall below that minimum speed even when you should. When it comes time to land you should be able to deploy flaps to not only slow down but lower you minimum flight speed (via increased lift). You should actually be able to slow down to any set speed and have the flaps deploy but that's more specifically targeted at only speed based flap deployment which is not solely what this is about.

We need a way to be able to go below the minimum airspeed on landing if flaps are deployed and even in some cases if they are not otherwise we truly never can fully benefit from them and slow the plane down and land as slow as you would like without risking setting the flight min speed too low. 

A couple ways to approach this:

• Allow FW_LND_AIRSPD_SC to go below 100% and only be active during the landing and flare portion of the flight.
• Speed based flap deployment
• Anything else anyone can think of :) 

Point is, we are limiting ourselves by not being able to utilize flaps to their full extent and this is something we see every time we come in to land, choose coming in faster than we like but having min fight speeds safe or lowering min fight speeds and coming in slower... Not a great tradeoff especially because in a lot of instances flaps (and full wing flaps) can really lower the landing speed a lot.

dagar, 2018-06-27:
What's the proper way to think about this fundamentally?

Ideas
 - flap deployment (variable) as a function of airspeed (commanded)
 - change minimum airspeed as a function of flap deployment

JonReacher, 2018-06-28:
> • Allow FW_LND_AIRSPD_SC to go below 100% and only be active during the landing and flare portion of the flight.

I've been thinking for a while that it would be nice to have a separate min airspeed for the land portion only, maybe even only activated at a max altitude above the land point to be safe.

ryanjAA, 2018-06-28:
Variable flap deployment as a function of commanded airspeed (ie speed based flaps) would be a great thing to add. There are some things done on the APM side already but i think it's better to approach it without looking how it's already been implemented to get a clear sense of what should happen.  We still need to separate flaps from airbrakes (crow) and we also need to think about landing cases where regardless of airspeed (although below some #) we want to deploy flaps to slow down and create lift to lower the min land speed.

Changing the min airspeed as a function of flap deployment could be good but how do we scale that? It would be highly airframe dependent as I do not see it being related linearly but maybe in practice that wont matter. Also we would need to think about mixers and flap servo range (and again also the airbrake flag).

@JonReacher Glad to hear that, this is definitely needed for the land portion, i'm not sure if you do it as a function of altitude but something like that (what happens is you have altitude error but are still landing and need to slow down to land but something, somewhere is the correct answer).

philipoe, 2018-06-28:
I think one of the main problems while descending during the landing approach is that we have the propeller to add energy into the system dynamically (i.e. through feedback terms in TECS) but we cannot take energy dynamically out of the system. The correct way to deal with this (on commercial aircraft) is that if the TECS throttle command is:

uThrot = 0...1 : Propeller adds energy into the system
uThrot = -1 ...0: Airbrakes/spoilers deploy proportionally to this value

This would be very effective during the landing. The question is only how that works in practice, as most small aircraft don't have airbrakes. They might have crow through the ailerons as @ryanjAA is mentioning, but the behavior of this is _very_ different from actual airbrakes (i.e. upwards ailerons will first dump lift _and_ actually decrease drag, whereas we'd be expecting braking action all the way through).

ryanjAA, 2018-06-28:
Some really good points. Can you explain how you envision scaling airbrakes/crow? I’d be concerned that absolute dynamic control at 0 throttle or less would lead to issues in wind where throttle from TECS may need to be enabled momentarily but that would make erratic changes to the flap/crow/airbrake deployment causing chaos if they were retracted suddeny. suppose we could add some logic to that to not disengage during that part of the descent. 

Definitely true on the lack of ability to take power out of the system easily. We do need reverse thrust for this but that’s not present here yet. 

As for ailerons up for crow, they definitely do decrease lift and drag on their own and but that’s not where the benefit comes. Two things that happen in practice are increased stability at slower speeds when you also have the flaps deployed and you are actually creating a vortex right where the flaps end and ailerons start which increases drag quite a lot to slow down. Due to the decreased lift on the outer span, wing tip vortices will be decreased some too (less drag but that is very much offset and assists reducing tip stall) but overall theory aside, drop the flaps, raise the ailerons and you have lower speed stability while slowing down and that’s all any of us have to work with right now (until reverse thrust comes into the codebase  :) ) so you’re exactly right, we need to command the deployment of flaps and/or crow accordingly. I think if we overthink this it might get a bit convoluted. Scaling flaps will work with airspeed. So long as we smooth that out not to retract them accidentally and we handle the crow logic.  

Putting a plan in place is possible but likely the best approach is to build this incrementally to see where holes show up. No matter what, we need to be able to go below TECS min speed or set TECS min speed lower based on flaps being deployed or simply if people want to have TECS min speed higher during normal flight for a safety margin. We need to deal with flaps and airbrakes/crow/etc being deployed during the landing phase which would also violate TECS min speed but we need to do so in a way where during landing we know we can deploy a crow setting if we want but not do so during normal flight. 

The takeaway so far:

- Flap deployment based on speed. Durning normal flight we need to segregate the airbrake call flag. 
- Flap deployment during landing with airbrake flag (I wouldn’t scale the flaps during landing but chime in here if you disagree). 
- TECS min speed logic. Need to be able to violate it during the landing phase and by a set amount/%.

philipoe, 2018-06-28:
> Some really good points. Can you explain how you envision scaling airbrakes/crow? I’d be concerned that absolute dynamic control at 0 throttle or less would lead to issues in wind where throttle [...]

One should _only_ set the airbrakes dynamically as a function of the (negative) throttle output by TECS. Flaps are not influenced by this. I then don't see any issues there. Airplanes by the way also dynamically (and quite quickly) engage/retract their speedbrakes. But again, whether this actually works nicely (i.e. creates the drag one might want during landing) depends on the specific plane configuration, so this requires a lot of testing.

ryanjAA, 2018-06-28:
One, I disagree and two, unless true standalone airbrakes are present (which are at the edge of a corner case) flaps are influenced from the perspective that there is no way to engage flaps without engaging the airbrakes and there are more than limited situations of true airbrake only users out there so we have to think about them in tandem and decouple the two surfaces. They are directly connected (as you probably know via) _att_sp.apply_flaps = true which engages flaperons which is how the "airbrakes" are enabled. Airbrakes should be set at the start of the landing sequence (if the user chooses to use them) which we never want to rely on negative throttle. We should already be at a predictable and set point at that time in the flight (and we are 99% of the time), if not violate mission landing logic and auto go around but I want to know whats going to happen in a mission and if I'm flying in an auto assisted mode and if we are trying to slow down and we have wind or are off slope, we will be pitching down and/or using throttle, less to manage the changing absolute flight characteristics, too many variables. I always find it funny that this is literally the most dangerous and difficult part of the entire flight process. You can set so many other aspects of the the firmware wrong and planes will just fly, a true testament to the hard work everyone has put in, tecs will compensate for pitch and throttle, all that but this is where it needs to be really robust.

Sure, maybe full-size craft do it (retract quickly) but what happens to managing quick changes in pitch nose up and down? You can tune that out without a problem but how long does that take and try explaining that to the average user. That has to be set precisely to avoid and that's the minority of people who have that correct but that's the issue with so much of this, we need to get away from what is done in other conventions and do what works, not as a hack but practically speaking. We want a robust system right? If we keep this only so granular that it requires the average user hours and hours of flight tests to use, the majority won't bother, or complain it never works and we shoot ourselves in the foot. Should granular control be possible? Absolutely but the code as is on the landing side (see #9778 ) needs much improvement we are far from perfect from how this works so starting small and building off that lets us all test what works and doesn't before committing to how everyone not testing will do it. 

We really need to think about this the following way, how do I give myself granular control if I want while creating something robust enough that it will work in the majority of cases without hours of minute testing and adding variables to manage in both auto and semi auto modes. Otherwise we sit stuck as a platform developers use and can make work but users get lost in.

JonReacher, 2018-07-01:
@dagar Since we already have reverse thrust in VTOL, would it be difficult to add the feature into FW? My typical AV is a flying wing, so reverse thrust (as opposed to flaps and other devices) is ideal for slower, steeper landing opportunities.

dagar, 2018-07-02:
@JonReacher it's not necessarily that difficult, but it will take a little bit of software work to get a clean architecture. The problem with the current VTOL reverse thrust implementation is that it's way too specific. Only applies to certain phases of standard VTOL where reverse can be activated via secondary channel.

Which type of ESC do you have that can reverse?

ryanjAA, 2018-07-02:
That's the other big thing. Type of esc. Since the castle units and the hobbywing ones do it (among all the simonk/blheli). The big thing is how they all do it differently and any lag. For instance the castle unit need to stop spinning prior to reverse once you hit the reverse channel which can cause a delay. The way around that is to turn on hard braking to make it instant but this either needs some parameters or when the higher powered zubax esc unit comes out since the current one doesn’t offer enough power for many fixed wings.

Kjkinney, 2018-07-02:
Could fixedwing reversed thrust be based off of re flashing an esc. to change it's midpoint? Or is that basically the same thing as using a channel? 

I think that's what ardupilot does as it says in their documentation to use a SimonK or Blheli supported esc.

JonReacher, 2018-07-02:
@dagar I currently have a Cobra ESC with BLheli firmware. It looks like the midpoint can be configured for reverse thrust. I can get whatever ESC is needed however, if not.

ryanjAA, 2018-07-02:
We’re onboard for testing and will get whatever needed/have most brands combos in stock here. Will help however possible. Yes. I agree. 8 deg to double would be great and with high powered motors you can go further than that. Plus we can basically eliminate ground roll with full reverse thrust once on the ground.

JonReacher, 2018-07-02:
Not to get too far ahead, but is there a list of reversible ESC's that we can come up with, or does any BLheli and/or SimonK firmware work?

ryanjAA, 2018-07-02:
Any blheli or simonk will do it. Castle creations will. Hobbywing new platinum series. The main thing to consider is how. Throttle midpoint or separate switch. And how we get either to work since there is the quality comcerm of the actual esc to consider when committing to implementing.

Antiheavy, 2018-07-13:
Don’t forget takeoff in this discussion. Flaps and other things may want to have specific and different behavior when in takeoff.

ryanjAA, 2018-07-13:
@antiheavy very very true and now with the recent flap pr and segregation of airbrake call we have much more to work with from a starting point but definitely does need to be part of this. We can already command pitch up prior to min speed but we need to think about exactly that, how we manage flaps accordingly on the way up.

philipoe, 2018-07-13:
> Don’t forget takeoff in this discussion. Flaps and other things may want to have specific and different behavior when in takeoff.

As @Antiheavy mentioned, new functionality for the takeoff flaps was merged through https://github.com/PX4/Firmware/pull/9858

junwoo091400, 2022-12-28:
Not sure but possibly dealt in the recent PR: https://github.com/PX4/PX4-Autopilot/pull/20443?

@KonradRudin @sfuhrer could you check on this? :thinking:

ryanjAA, 2022-12-28:
Yes, it is dealt with. You can go below min speed and not trigger under speed now. Still testing but will re-open if necessary.

closed 2022-12-28 by ryanjAA