PX4/PX4-Autopilot #9778: Auto-landing Methodology

ryanjAA, 2018-06-27:
Having speaking with @dagar quite extensively and then our Aero guys here, it's clear that while with some serious tweaking, auto landing can be used very reliably (we were in 30+ mph winds the other day and all was repeatedly fine), it takes a lot of effort to get autolanding not only tuned properly, but also robustly. After much deliberation, it comes down to the current implementation and the methodology behind it should be altered.

The top 2 reasons auto-landing doesn't work well is speed and ground detection. Ground detection is more straight forward, use a lidar (but even that's overly simplistic as altitude estimates need to be fused but hold that constant for a bit, i.e. terrain estimator is too complicated). Speed is the second and predominant issue. Too much speed messes with landings because your flare all of a sudden is now actually a slight takeoff, altitude gaining event.

 Two main ways to look at landing:

• Navy carrier landings where you essentially flare to stop your decent, and allow landing gear to absorb the impact. 
Or 
• Flare like a real plane. 

Flaring like a real plane is the ideal path otherwise the landing gear runs the risk of causing the plane to bounce into the air, etc, break, cause repeated stress, etc. 

The issue currently is that a flare isn’t a preset numerical value. 

On approach the aircraft should be slowing, even though it is descending. As it enters ground effect (in our case, about 0-6ft high) there is substantial compression lift underneath the plane, this essentially means increased performance. In other words, it can fly and take off significantly slower while in ground effect, than it can say at 40 ft high. Because of this, a gradual and steady approach is flown, again reducing speed until pretty much over the end of the runway. As an aircraft passes the end of a runway it is standard to pull power back to idle and standard procedure is to “float” in ground effect and gradually pull back on the stick more and more as the speed slows, and the aircraft loses lift. The text book landing will stall the wing as the main landing gear touch, and keep the nose off the ground until the tail no longer has the aerodynamic authority to keep it in the air. This means the aircraft can not bounce, or fly. We have onboard footage of this during our tuft testing. The tufts indicate a stall and go wild just as the aircraft touches the runway.

Flare is not a one step maneuver, it is a gradual and progressive increase in angle of attack. A balance between lift and airspeed, all while trying to time and calculate your impact with the ground. If it is looked at as a simple 5 degree sudden pitch change, it will be impossible to have smooth landings.

A perfect landing with an autopilot can be fairly difficult unless you can set lidar (or altitude estimate) to literally a few inches off the ground and tell it to maintain separation. At which point the autopilot should command increasing angle of attack as the aircraft slows until it eventually stalls and sticks the landing. 

The best way to do this would be to fly a slower approach without stalling and utilizing crow (see #9777 ). When over the runway the aircraft can be commanded to maintain 6 inches off the ground by increasing angle of attack, while speed drops. Upon contact with the ground we can do 3 things that will help it not bounce. 
• Elevator neutral (always) 
• Retract flaps immediately (dumping lift).
• If possible, reverse the motor

I think that we need to rethink the landing code to at least in broad-strokes take the above into account. Thoughts or ideas are always helpful for the discussion here but there is some serious room for improvement that really would just make things much more robust on the landing side of things for any airframe. 

Video of tufts at landing (https://youtu.be/sRGfdQSMwow?t=156)

dagar, 2018-06-27:
Thanks for sharing, let's try to turn this into something actionable.

Detecting ground contact could be interesting for properly switching over to the wheel controller.

philipoe, 2018-07-23:
> I think that we need to rethink the landing code to at least in broad-strokes take the above into account. 

I'd agree that the "how to flare" code is suboptimal and needs rework. The "approach" code is not too bad in my opinion, at least if you are ok with NOT using spoilers/crow/reverse motor...

closed 2019-02-04 by stale[bot]

ryanjAA, 2019-02-26:
This would be good to re-look at. @antiheavy, any thoughts would be appreciated.

Antiheavy, 2019-03-02:
@ryanjAA maybe move discussion to PX4_Discuss?  http://discuss.px4.io/t/fixed-wing-auto-landing-brainstorm/9837