PX4/PX4-Autopilot #9782: Manual control and rc override in SITL not working

Torgenn, 2018-06-27:
Sending manual control or rc override commands to the SITL through Mavros does not seem to work. 

I posted a message on the forum ([http://discuss.px4.io/t/sitl-remote-control-simulation/6993](http://discuss.px4.io/t/sitl-remote-control-simulation/6993) ) that has more details. I found other people having a similar problem with rc override, but no real answer. I am looking for information to know if it is indeed not enabled in the SITL or if I'm doing something wrong.

I have attached the code from my node for information.
[rc_simulator_node.txt](https://github.com/PX4/Firmware/files/2142461/rc_simulator_node.txt)
Thanks

TSC21, 2018-07-04:
https://github.com/PX4/Firmware/blob/master/src/modules/simulator/simulator_mavlink.cpp does not handle [`RC_CHANNELS_OVERRIDE`](https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE). That's why it's not processing them. If you do want it supported, you need to add the handler there.

mhkabir, 2018-07-05:
Actually, no. Whatever you send through Mavros is processed independent of `simulator_mavlink`. The simulator module only handles communication between the simulator (e.g Gazebo) and PX4 executable.

The standard mavlink module (`mavlink_receiver`) handles communications with Mavros. https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1676-L1732.

closed 2018-07-05 by mhkabir

TSC21, 2018-07-05:
@mhkabir right but then that means this issue is not solved, unless @Torgenn is making some mistake on the ROS side.

pmdoll, 2019-09-06:
Do we have a resolution to this issue? A working example perhaps.