PX4/PX4-Autopilot #9783: frsky_telemetry using hardware flow control com port on pixhawk (solution)

gianni-carbone, 2018-06-27:
Using frsky_telemetry (D TYPE) with output on a port with hw flow control (i.e. /dev/ttyS1) does not work. 
Attach frsky_telemetry with frsky_telemetry start -d /dev/ttyS1 to a type D receiver it produces "INFO [frsky_telemetry] sending frSky D type telemetry" but no data flows on the channel.
Notice: In tested configuration mavlink telemetry was stopped at /dev/ttyS1 before frsky device driver started.
The needs of using ttyS1 due to SYSCOMPANION = mavesp and serial 4 used for tfmini ranger. This way only serial line available is ttyS1 on pxhawk hardware.

Solution:
Add this lines to frsky_telemetry.cpp SOLVE the problem
... just after:
	/* Disable output post-processing */
	uart_config->c_oflag &= ~OPOST;
..in static int sPort_open_uart
.. add:

	uart_config->c_cflag |= (CLOCAL | CREAD);    /* ignore modem controls */
	uart_config->c_cflag &= ~CSIZE;
	uart_config->c_cflag |= CS8;         /* 8-bit characters */
	uart_config->c_cflag &= ~PARENB;     /* no parity bit */
	uart_config->c_cflag &= ~CSTOPB;     /* only need 1 stop bit */
	uart_config->c_cflag &= ~CRTSCTS;    /* no hardware flowcontrol */

	/* setup for non-canonical mode */
	uart_config->c_iflag &= ~(IGNBRK | BRKINT | PARMRK | ISTRIP | INLCR | IGNCR | ICRNL | IXON);
	uart_config->c_lflag &= ~(ECHO | ECHONL | ICANON | ISIG | IEXTEN);

	/* fetch bytes as they become available */
	uart_config->c_cc[VMIN] = 1;
	uart_config->c_cc[VTIME] = 1;

bkueng, 2018-06-28:
Hi @gianni-carbone 
Thanks for investigations. Can you create a pull request?

`uart_config->c_cc[VMIN] = 1;`
I'm not sure about this line, it could lead to early `read` returns, with only a few bytes, but the parser expects to read a full packet.

gianni-carbone, 2018-06-28:
Hi @bkueng 
At the moment:
uart_config->c_cc[VMIN] = 1;
uart_config->c_cc[VTIME] = 1;
does work with no problem.
I will try w/out it. If ok I will create a pull request.
I would refactor just some line of code too, just little thinghs like:
using errx(0, "running: ... instead of PX4_INFO(" ... etc.

bkueng, 2018-06-28:
> I will try w/out it. If ok I will create a pull request.

Sounds good.

> I would refactor just some line of code too, just little thinghs like: using PX4_INFO(" running: ... in place of errx(0, "running: ... etc.

Even better, we want to get rid of `errx`

gianni-carbone, 2018-06-29:
Hi @bkueng,
just cheked suppressing uart_config->c_cc[VMIN] = 1; and uart_config->c_cc[VTIME] = 1;
it works ok.
Added some refactors, added a sent packet counter and refactor the "status" section. With this the "status" command shows the sent packet counter and the phisical port attached to.

commit 16d67ed1 (2018-06-29): frsky_telemetry improved com port init (fix #9783), minor refactor (#9798)

* frsky_telemetry added support for hw flow control com port and minor improvements

* DTYPE tested OK. Return 0 for unix compatibility, whitespace removed. Full test start - status- stop - start -status OK
closed 2018-06-29 by LorenzMeier