PX4/PX4-Autopilot #9787: Can't get SITL working on 1.8 update

JonReacher, 2018-06-28:
ends with
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

works fine on 1.7.3

hamishwillee, 2018-06-28:
Host platform you are building on? Can you add a full build log as a Gist and link?

JonReacher, 2018-06-28:
Building on mac. 
Gist: https://gist.github.com/JonReacher/58b0df3580aa29989f2b1cef36e6e1f8#file-gistfile1-txt

hamishwillee, 2018-06-28:
Not an expert, but possibly  missing datasets.

Then install [GeographicLib](https://geographiclib.sourceforge.io/) datasets by running the `install_geographiclib_datasets.sh` script:

I'm still getting doc update reviewed: https://github.com/PX4/Devguide/pull/526

LorenzMeier, 2018-06-28:
Did you run ‘make distclean’ after the update to 1.8?

JonReacher, 2018-06-28:
I did run make distclean first thing. How exactly do I install the datasets?  I have tried to look up apt-get, etc., but am getting nowhere. Are there any very clear and concise, 'do this' type of instructions anywhere? I'm not a coder but I need to run simulations before putting expensive AVs in the air on updated software. That's bitten me before.

Do I need to upgrade from Gazebo 7.9?

JonReacher, 2018-06-28:
@hamishwillee Finally got Geographiclib installed, but no difference.

TSC21, 2018-06-28:
@JonReacher did you follow https://dev.px4.io/en/setup/dev_env_mac.html?

hamishwillee, 2018-06-28:
@JonReacher As @TSC21 says above https://github.com/PX4/Firmware/issues/9787#issuecomment-401199818 , that should work out of the box. If not and you have a clean Firmware installation, the problem is probably our side (you should not have to update Gazebo).

FYI, we in theory support Gazebo 7, 8, and 9. I would have thought the MacOS installer would do 8.

TSC21, 2018-06-28:
> FYI, we in theory support Gazebo 7, 8, and 9. I would have thought the MacOS installer would do 8.

We have CI building for MacOS for `sitl_gazebo`, and it does not complain about the above error. So I suppose you are missing something on your toolchain @JonReacher.

TSC21, 2018-06-28:
@JonReacher please do follow https://dev.px4.io/en/setup/dev_env_mac.html. It will install what you required on your end. It will also install Gazebo 8 on your system. No need to install `Geographiclib` or its datasets for this, unless you want to use MAVROS.

JonReacher, 2018-06-29:
@TSC21 I followed the guide and reinstalled everything on the linked page. Now I have this:

#error This file was generated by an older version of protoc which is
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/material.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

TSC21, 2018-06-29:
@JonReacher I don't own a Mac myself, but I would say that you have incompatible packages installed on your system.

JonReacher, 2018-06-30:
@TSC21 I un/reinstalled everything in the setup guide. I also un/reinstalled the latest Gazebo version. 

Here's what I have now:

CMake Error in CMakeLists.txt:
  Imported target "dart" includes non-existent path

    "/usr/local/Cellar/libccd/2.0_2/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
-- Build files have been written to: /Users/jonreacher/JRsrc/Firmware/build/posix_sitl_default/build_gazebo
[388/636] Building CXX object src/modu...es__mavlink.dir/mavlink_receiver.cpp.o
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

@hamishwillee- as a side note, the Gazebo sim guide mentions that "Gazebo 8 setup is included in our standard build instructions", but I don't see that in the instructions.

hamishwillee, 2018-06-30:
> @hamishwillee- as a side note, the Gazebo sim guide mentions that "Gazebo 8 setup is included in our standard build instructions", but I don't see that in the instructions.

What about the first paragraph in https://dev.px4.io/en/setup/dev_env_mac.html ?

>  The following instructions explain how to set up a development environment for building NuttX-based hardware (Pixhawk, etc.) and Simulation (jMAVSim/**Gazebo8**) targets.

JonReacher, 2018-06-30:
@hamishwillee I installed Gazebo myself as a separate install. Does that mean that's not necessary, or even should not be done because it's already done by following the steps given?

IngTIKNA, 2023-05-28:
It was the same issue I encountered while building the px4_sitl in the virtual conda environment. It has gone after I have deactivated the virtual environment. I was encountering the same issue during building the px4_sitl while the virtual conda environment was active.

julianoes, 2023-05-29:
This issue is ancient. @IngTIKNA please don't necrobump and create a new issue where you properly describe your problem.