PX4/PX4-Autopilot #9790: Pixhawk 4, EKF High IMU accel bias. (arming very difficult)

jstrebel, 2018-06-28:
Hello, I tried to put a new Holybro 4 with 1.8.0 into my fixed wing (using the GPS and Magnetometer from the Holybro 4)
Before was holybro PIX32 successful in place in the same UAV. 

I get  “critical: preflight fail:EKF High IMU accel bias)” and “Critical: Not arming: Preflight checks failed”
Tried several times the following:
Run power the unit for 30min and started the calibriation from “ground up” again.
Took a second Holybro 4 Unit and tried again. Tried to change the param: COM_ARM_MAG from default to 0.5, did not change anything.
Up to now, I was only able to go once successful trough the calibration process. Out of then attempts. Calibration is done outside a building, 
What is your advice to get to the root cause? Using QGC daily built from 22-6
Thanks for your help. Jakob
 
https://logs.px4.io/plot_app?log=2e9582c4-95c6-43ca-88fb-a388eff98028

AlexisTM, 2018-06-28:
Hey Jacob,
What I see in the logs is many "Manual control lost: failsafe" can you check your remote control connection?

jstrebel, 2018-06-28:
@AlexisTM Alexis, My RC was not switched on during calibration.  Thx Jakob

jstrebel, 2018-07-01:
After trying unsuccessfully to calibrate the Pixhawk 4 countless times via QGC, I did a param reset via console on the pixhawk. 
Heureka: the error message is gone.
I already did a parameter reset via QGC before that. But it seems that this did not have the same effect.
My suspicion is that because I work with several different pixhawk HW controllers, something has got mixed up. Hopefully this info helps other to save some time. Jakob

AlexisTM, 2018-07-02:
Good tip! I will try this as well!

Antiheavy, 2019-01-22:
related: https://github.com/PX4/Firmware/issues/10833

closed 2019-07-09 by stale[bot]

AlexisTM, 2019-07-09:
I had the High IMU accel bias problem. It was caused by the onboard computer fan that was too close of the Pixhawk and not dampened.