PX4/PX4-Autopilot #9793: mc_pos_control: switch to use bottom distance when stationary; when moving horizontally use local position

AuterionWrikeBot, 2018-06-28:


AuterionWrikeBot, 2018-06-28:
➤ Thomas Gubler commented:

@Paul Riseborough this is a subtask of https://www.wrike.com/open.htm?id=232565644 (but it was not assigned to you in wrike)

AuterionWrikeBot, 2018-06-28:
➤ Thomas Gubler commented:

As discussed with Paul: MC changes conflict with flight tasks. He will work on top of flight tasks branch ( https://github.com/PX4/Firmware/pull/9731 )

ndepal, 2018-06-29:
It sounds like this is already achieved with `EKF2_RNG_AID`? Or does this add other functionality?

priseborough, 2018-07-07:
I have raised a PR for this: https://github.com/PX4/Firmware/pull/9854

ndepal, 2018-07-09:
I still don't understand the difference, both `EKF2_RNG_AID` and this seem to use range data when not moving, and the primary height source otherwise.

I hope the documentation will make the difference between the two clear.

priseborough, 2018-07-09:
OK, I can see how that is confusing.

EKF2_RNG_AID was originally developed for an application where the ground effect interference from rotor wash was so severe that i t affected vertical velocity estimates and  safe takeoffs and landings were not possible. Range finders are accurate, but not reliable sensors (they can experience large random readings), the terrain under the vehicle cannot be relied on to be flat and although the EKF can handle slow drift and variation in height reference, anything more than that can cause estimation problems that go beyond the vertical position state. For these reasons, unless baro errors are severe, we should be meeting the requirement for accurate ground relative height control  in the position controller when possible.

The EKF2_RNG_AID parameter description has been updated in the PR to make this clearer.

ndepal, 2018-07-10:
Got it, thanks for the explanation and the updated description!

thomasgubler, 2018-07-20:
TEST bump for bot, disregard

priseborough, 2018-07-23:
Now in master

closed 2018-07-23 by priseborough