PX4/PX4-Autopilot #9796: IMU message for vision

RomanBapst, 2018-06-28:
There are currently two IMU messages which can be streamed via mavlink to a companion computer which runs a vision algorithm. At the moment I think none of them is really suited due to the following reasons:

1) HIGHRES_IMU
- these are the imu values from the sensor_combined topic
- the gyro & accel values are low pass filtered which might not be optimal for vision
- sensor_combined does contain unfiltered delta velocity and delta angle data which would be suited for estimation but the vision algorithms normally don't support this

2) SCALED_IMU
- these should be raw values (no filtering) and ideal for vision
- for any other platform than Snapdragon the values written to the message are wrong (this is because Snapdragon uses a different driver which does the computation correctly). The mavlink module assumes that the fields x_raw, y_raw, z_raw in the sensor_accl & sensor_gyro topic have units m/s² and rad/s respectively. However, the native Nuttx drivers do not apply the necessary scaling and so the assumption is wrong. It's also not very helpful that the orb message does not indicate what units these fields should have, see https://github.com/PX4/Firmware/blob/master/msg/sensor_accel.msg#L13
- the sensor data of this topic is represented in the board frame. There is not way the companion computer side can figure out the board rotation. The values should be rotated according to the board rotation before being sent out.

**How to fix this**
- first of all we should add documentation to the orb message indicating the units of the raw imu data
- we need to rotate the imu data when preparing the SCALED_IMU message (both board rotation and board rotation offset)
- we either need to fix the NuttX drivers or the mavlink module (depending on what units the x/y/z_raw fields will have) so that the SCALED_IMU message does not contain garbage.

- Alternatively we could argue to use the HIGHRES_IMU message. In that case we need to add unfiltered sensor data to it. It actually already contains unfiltered delta velocities and delta angles which would be suited for estimation, however, if the vision algorithms do not support these then they are useless.

@LorenzMeier @mhkabir FYI

RomanBapst, 2018-06-28:
If I can get some  opinions on what the best way would be then I can go ahead and provide the fix.

mhkabir, 2018-06-28:
> sensor_combined does contain unfiltered delta velocity and delta angle data which would be suited for estimation but the vision algorithms normally don't support this.

This is not strictly true. The way the MAVLink sender for HIGHRES_IMU divides the delta angle/velocity by the integration time gives a rate and acceleration which is fine for external estimators. 

Have a look at this branch which I'm using for VIO : https://github.com/mhkabir/Firmware/commits/artemis-racer-test3

The important commits are the ones where I fixed the sensor driver timestamping to take into account the on-chip lowpass delay. In my setting, this was all that was needed to get really good VIO performance with aggressive flight. https://github.com/mhkabir/Firmware/commit/ee5bb2c174314324acd1c8676bb7004495a350cd

I vote to fix the time-stamping in a nice way and evaluate performance of VIO after that. We should not even need the SCALED_IMU stuff. Would like to discuss this over voice if you have time @RomanBapst.

mhkabir, 2018-06-28:
Combined with correct temperature calibration for the IMU and my branch, VIO performance is already as good as with much higher-end IMUs. Ex.: https://www.youtube.com/watch?v=cFKByCyX0k0.

LorenzMeier, 2018-06-28:
@mhkabir That looks pretty nice and is something we need anyway for the overall system architecture (accounting for correct sensor latencies).

mhkabir, 2018-06-28:
I can actually do a batch optimization based time alignment using angular velocities from a feature tracker and the IMU angular velocities to see the accuracy of our timestamping. Should be worth it.

RomanBapst, 2018-06-28:
@mkabir ok, we can have a call. I'll also add Leonie into the loop, she is leading these efforts.

mhkabir, 2018-06-28:
Cool, let's coordinate over Slack.

LorenzMeier, 2018-06-29:
I forgot one aspect: Vibration. The time delay compensation only works if both the exposure time and the low pass filter have similar integration times (and hence similar frequency responses). Otherwise we will see aliasing when fusing the result. I would over-do it right now but quickly calculate the frequency response of a box filter with typical exposure times (0.1 ms outdoors and closer to 1 ms indoors). The IMU filter should not have a lower cutoff than the box filter.

That alone can make a separate data path for inertial fusion mandatory for optimal performance.

dagar, 2018-10-03:
Added to the Driver Architecture / IMU Signal Processing project board. https://github.com/PX4/Firmware/projects/23

closed 2019-02-04 by stale[bot]