PX4/PX4-Autopilot #9803: [Bebop 2] Ekf2 doesn't initialise, don't get a baro selection

bramsvs, 2018-06-29:
**Describe the bug**
The ekf2 estimator fails to initialise, baro sensor is not selected.

**To Reproduce**
Steps to reproduce the behavior, on latest master:
1. Do a posix_bebop_default upload, need to fix the `../Tools/adb_upload.sh` location.
2. Run PX4.

**Log Files and Screenshots**
Log file: https://review.px4.io/plot_app?log=c1b21c29-6390-460a-9533-fed177b1bf60

```
pxh> listener sensor_selection

TOPIC: sensor_selection instance 0 #1
 sensor_selection_s
        timestamp: 344269 (21.348424 seconds ago)
        accel_device_id: 4548616
        baro_device_id: 0
        gyro_device_id: 4548616
```

bramsvs, 2018-06-29:
See also this PR: https://github.com/PX4/Firmware/pull/9805 to get the upload script going again!

mhkabir, 2018-06-29:
Is the baro driver functioning? Try `listener sensor_baro`.

bramsvs, 2018-06-29:
@mhkabir Will check coming Monday! :)

bramsvs, 2018-07-02:
This is what I get:
```
pxh> listener sensor_baro

TOPIC: sensor_baro instance 0 #1
 sensor_baro_s
        timestamp: 92267170 (0.014974 seconds ago)
        error_count: 4033114797442235
        pressure: 101889.000
        temperature: 44.500
        device_id: 28893088
```
Also did a `sensors status`:
```
pxh> sensors status
INFO  [sensors] gyro status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 75, state: OK
INFO  [ecl/validation]        val:   0.0039, lp:   0.0005 mean dev:   0.0000 RMS:   0.0080 conf:   1.0000
INFO  [ecl/validation]        val:   0.0015, lp:   0.0002 mean dev:   0.0000 RMS:   0.0077 conf:   1.0000
INFO  [ecl/validation]        val:   0.0012, lp:  -0.0000 mean dev:  -0.0000 RMS:   0.0011 conf:   1.0000
INFO  [sensors] accel status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 75, state: OK
INFO  [ecl/validation]        val:   0.3253, lp:   0.2810 mean dev:  -0.0002 RMS:   0.0436 conf:   1.0000
INFO  [ecl/validation]        val:  -0.3701, lp:  -0.3375 mean dev:   0.0007 RMS:   0.0782 conf:   1.0000
INFO  [ecl/validation]        val: -20.2250, lp: -20.3451 mean dev:   0.0019 RMS:   0.2683 conf:   1.0000
INFO  [sensors] mag status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 100, state: OK
INFO  [ecl/validation]        val:  -0.1925, lp:  -0.1899 mean dev:  -0.0000 RMS:   0.0080 conf:   1.0000
INFO  [ecl/validation]        val:   0.0925, lp:   0.0994 mean dev:  -0.0000 RMS:   0.0075 conf:   1.0000
INFO  [ecl/validation]        val:   0.3750, lp:   0.3721 mean dev:   0.0003 RMS:   0.0079 conf:   1.0000
INFO  [sensors] baro status:
INFO  [ecl/validation] validator: best: -1, prev best: -1, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 75, state: ECNT
INFO  [ecl/validation]        val: 101879.0000, lp: 101884.4453 mean dev:  -1.0167 RMS:   5.9607 conf:   0.0000
INFO  [ecl/validation]        val:  44.4900, lp:  44.4913 mean dev:   0.0038 RMS:   0.0100 conf:   0.0000
INFO  [ecl/validation]        val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   0.0000
INFO  [sensors] Temperature Compensation:
INFO  [sensors]  gyro: enabled: 0
INFO  [sensors]  accel: enabled: 0
INFO  [sensors]  baro: enabled: 0
INFO  [sensors] Airspeed status:
INFO  [ecl/validation]        no data
```

There is a high error count, weird thing is that the error count doesn't increase after starting px4, it just remains constant. I will check if I can test it on another Bebop to see what happens.

@mhkabir do you have PX4 v1.8.x ekf2 initializing correctly on a Bebop? It's unclear to me what the state of Bebop support is at this moment, see: https://github.com/PX4/Firmware/issues/7258#issuecomment-302154275 so I'd love if someone can confirm its working/broken condition!

mhkabir, 2018-07-02:
There is a bunch of stuff broken like IMU drivers which need to be updated to send integral values for the estimator, and filtered values for the rate controllers and such.

Baro might be broken as well, but it was working 1-2 releases back.

RomanBapst, 2018-07-02:
@bramsvs The last time I used the Bebop2 was at the beginning of 2017. I had it at the point where it flew very nicely in position control and you could fly missions. Unfortunately, some time later there were a few changes made to the rate control path (multirotors subscribed directly to the raw imu messages) and I think it must have broken around that time. I never tried it after this and never heard of anybody flying it after. I wanted to test it about a month ago but my battery is flat :-(
If you need help on this you can let me know.
The first thing I would do is to check if all the sensor topic are published, you can do that using:
````
listener sensor_gyro
listener sensor_accel
listener sensor_mag
listener sensor_baro
````

bramsvs, 2018-07-02:
I did run the Bebop in manual mode while holding in my hand. It is certainly controlling roll/pitch/yaw rates when giving inputs or rotating it so I'm hopeful about the attitude rate control loop. @mhkabir Isn't the rate filtering happening in `mc_att_control_main` itself?

@RomanBapst Do these look okay(ish)?  
```
pxh> listener sensor_accel

TOPIC: sensor_accel instance 0 #1
 sensor_accel_s
        timestamp: 169002981 (0.004862 seconds ago)
        integral_dt: 3835
        error_count: 0
        x: 0.242
        y: -0.313
        z: -20.661
        x_integral: 0.001
        y_integral: -0.001
        z_integral: -0.079
        temperature: 0.000
        range_m_s2: -1.000
        scaling: -1.000
        device_id: 4548616
        x_raw: 0
        y_raw: 0
        z_raw: 0
        temperature_raw: 0

pxh> listener sensor_gyro

TOPIC: sensor_gyro instance 0 #1
 sensor_gyro_s
        timestamp: 175562216 (0.001305 seconds ago)
        integral_dt: 3666
        error_count: 0
        x: 0.001
        y: 0.010
        z: 0.002
        x_integral: 0.000
        y_integral: 0.000
        z_integral: 0.000
        temperature: 0.000
        range_rad_s: -1.000
        scaling: -1.000
        device_id: 4548616
        x_raw: 0
        y_raw: 0
        z_raw: 0
        temperature_raw: 0

pxh> listener sensor_mag

TOPIC: sensor_mag instance 0 #1
 sensor_mag_s
        timestamp: 178528473 (0.000949 seconds ago)
        error_count: 0
        x: 0.037
        y: -0.144
        z: 0.377
        range_ga: 0.000
        scaling: 0.000
        temperature: 0.000
        device_id: 4590856
        x_raw: 0
        y_raw: 0
        z_raw: 0
        is_external: 1

pxh> listener sensor_baro

TOPIC: sensor_baro instance 0 #1
 sensor_baro_s
        timestamp: 182425600 (0.003359 seconds ago)
        error_count: 4015625690879163
        pressure: 101818.000
        temperature: 34.350
        device_id: 15748976
```
It looks like it isn't doing any accelerometer calibration..

I will try to get another Bebop on Wednesday and see what happens to the baro errors.

Thanks all for your help! Would be nice if we can get this thing going again :)

bramsvs, 2018-07-04:
The problem was the error count, see: https://github.com/PX4/Firmware/pull/9835. Haven't flown it yet but the ekf initializes! :) 

It is maybe a good idea to make a general Bebop 2 issue tracking issues related to the Bebop.. Some of the problems I ran into:
* Upload didn't work (open PR)
* This baro error_count issue (open PR)
* Accelerometer calibration is way off
* Don't get rangefinder output
* The cpu/board fan doesn't run (you can start it by running `DragonStart.sh &` and then killing it with `kk`)

Other broken/almost done features:
* Motor rpm output
* Downward looking optical flow camera
* Forward looking camera

@eyeam3 It seems you're listed as Bebop 2 maintainer.. Any chance you will pick this task up again? Or do we need someone else? I'm happy to test/fix stuff but I'm not really an embedded software guy (yet).

eyeam3, 2018-07-09:
@bramsvs I worked on the PX4-Bebop setup at the very beginning, but haven't been working on it for more than a year now. Unfortunately, I will not pick this task up again, so there should be someone else assigned as the maintainer.

asimonov, 2018-07-22:
I have similar issues with baro. here is my log:

```
/ # /data/ftp/internal_000/px4 /home/root/px4.config 
commands file: /home/root/px4.config

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [param] selected parameter default file /home/root/parameters
139446 IMU temperature initialized to: 50.344704
172326 Initialize device: /sys/bus/iio/devices/iio:device0
179088 Initialize device: /sys/class/pwm/pwm_9
179720 Initialize device: /sys/class/gpio/gpio129
INFO  [bebop_flow] Set image format: 320x240
    Format: NV12
    Size: 115200
INFO  [bebop_flow] Streaming started: /dev/video0
WARN  [commander] LED: getHandle fail

WARN  [commander] LED init failed
INFO  [dataman] Unknown restart, data manager file '/data/ftp/internal_000/fs/microsd/dataman' size is 11405132 bytes
INFO  [ekf2] Found range finder with instance 0
Sleeping for 1 s; (1000000 us).
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
Sleeping for 1 s; (1000000 us).
INFO  [df_bebop_bus_wrapper] Trying to initialize mixers from config file /home/root/bebop.main.mix
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log
INFO  [logger] Opened log file: /data/ftp/internal_000/fs/microsd/log/sess004/log001.ulg
pxh> INFO  [mavlink] using network interface eth0, IP: 192.168.42.1
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 192.168.42.255
INFO  [mavlink] partner IP: 192.168.42.47
WARN  [commander] Preflight checks always pass on Bebop.
WARN  [commander] Preflight checks always pass on Bebop.
WARN  [commander] Preflight checks always pass on Bebop.
WARN  [commander] Preflight checks always pass on Bebop.
WARN  [commander] Preflight checks always pass on Bebop.
WARN  [commander] Preflight checks always pass on Bebop.
WARN  [commander] Preflight checks always pass on Bebop.
INFO  [commander] data link #0 lost
WARN  [commander] Preflight checks always pass on Bebop.
INFO  [commander] data link #0 regained

pxh> 
pxh> 
pxh> listener sensor_selection

TOPIC: sensor_selection instance 0 #1
 sensor_selection_s
        timestamp: 351096 (56.904138 seconds ago)
        accel_device_id: 4548616
        baro_device_id: 0
        gyro_device_id: 4548616
        mag_device_id: 4590856
pxh> listener sensor_baro

TOPIC: sensor_baro instance 0 #1
 sensor_baro_s
        timestamp: 79765579 (0.002616 seconds ago)
        error_count: 4073951345874107
        pressure: 101717.000
        temperature: 38.430
[... 78 lines cut ...]
 sensor_mag_s
        timestamp: 238559930 (0.000820 seconds ago)
        error_count: 0
        x: 0.182
        y: -0.123
        z: 0.418
        range_ga: 0.000
        scaling: 0.000
        temperature: 0.000
        device_id: 4590856
        x_raw: 0
        y_raw: 0
        z_raw: 0
        is_external: 1
pxh> listener sensor_baro

TOPIC: sensor_baro instance 0 #1
 sensor_baro_s
        timestamp: 254225477 (0.001202 seconds ago)
        error_count: 4073951345874107
        pressure: 101735.000
        temperature: 42.140
        device_id: 12914104
pxh> 
```