PX4/PX4-Autopilot #9807: bebop2: libstdc++.so.6 version GLIBCXX_3.4.21 not found

asimonov, 2018-06-29:
**Describe the bug**
Final px4.stripped executable does not work on Parrot Bebop 2 (installed on top of latest Parrot firmware 4.7.1)

**To Reproduce**
Steps to reproduce the behavior:
1. Build 1.8.0 for bebop 
2. Upload to the drone
3. run 'px4.stripped /home/root/px4.config'

gives error:

/lib/libstdc++.so.6: version 'GLIBCXX_3.4.21' not found (required by px4.stripped)

**Expected behavior**
runs ok

**Drone (please complete the following information):**
Parrot Bebop 2

Using latest cross-compile tools on Ubuntu:
```
$ dpkg -l | grep armhf
ii  cpp-arm-linux-gnueabihf                                     4:5.3.1-1ubuntu1                                      amd64        GNU C preprocessor (cpp) for the armhf architecture
ii  g++-arm-linux-gnueabihf                                     4:5.3.1-1ubuntu1                                      amd64        GNU C++ compiler for the armhf architecture
ii  gcc-arm-linux-gnueabihf                                     4:5.3.1-1ubuntu1                                      amd64        GNU C compiler for the armhf architecture
ii  libasan2-armhf-cross                                        5.4.0-6ubuntu1~16.04.9cross1                          all          AddressSanitizer -- a fast memory error detector
ii  libatomic1-armhf-cross                                      5.4.0-6ubuntu1~16.04.9cross1                          all          support library providing __atomic built-in functions
ii  libc6-armhf-cross                                           2.23-0ubuntu3cross1                                   all          GNU C Library: Shared libraries (for cross-compiling)
ii  libc6-dev-armhf-cross                                       2.23-0ubuntu3cross1                                   all          GNU C Library: Development Libraries and Header Files (for cross-compiling)
ii  libgcc-5-dev-armhf-cross                                    5.4.0-6ubuntu1~16.04.9cross1                          all          GCC support library (development files)
ii  libgcc1-armhf-cross                                         1:5.4.0-6ubuntu1~16.04.9cross1                        all          GCC support library
ii  libgomp1-armhf-cross                                        5.4.0-6ubuntu1~16.04.9cross1                          all          GCC OpenMP (GOMP) support library
ii  libstdc++-5-dev-armhf-cross                                 5.4.0-6ubuntu1~16.04.9cross1                          all          GNU Standard C++ Library v3 (development files)
ii  libstdc++6-armhf-cross                                      5.4.0-6ubuntu1~16.04.9cross1                          all          GNU Standard C++ Library v3
ii  libubsan0-armhf-cross                                       5.4.0-6ubuntu1~16.04.9cross1                          all          UBSan -- undefined behaviour sanitizer (runtime)
ii  linux-libc-dev-armhf-cross                                  4.4.0-18.34cross1                                     all          Linux Kernel Headers for development (for cross-compiling)
```

bramsvs, 2018-06-29:
Did you do this https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html#raspberry-pi-hardware ?

mhkabir, 2018-06-29:
You need an older crosscompiler. The last version which works for the currently-shipping Bebop linux system is `gcc-linaro-4.9.4-2017.01`

hamishwillee, 2018-07-02:
@mhkabir In https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html#raspberry-pi-hardware we recommend GCC4.8.3:

>  The recommended toolchain for raspbian in our docs is GCC 4.8.3 and can be cloned from https://github.com/raspberrypi/tools.git

Looking at that git repo, I think we need to change this to gcc 4.9.3  - correct?

mhkabir, 2018-07-02:
Yes, we can change it to `gcc-linaro-4.9.4-2017.01`, but it is not strictly necessary. It's just the last version I found to work with the Bebop's current OS image. Also I have not validated this version on the Pi, although it will very likely work.

hamishwillee, 2018-07-02:
@mhkabir Thanks, I have updated https://github.com/PX4/Devguide/pull/552 to make it clear that the "current correct" toolchain is in the linked repo (irrespective of what that toolchain is)

asimonov, 2018-07-02:
thanks! installing raspberry pi crosscompile tools solved the issue. using gcc 4.8.3 i can build and run code on parrot bebop

closed 2018-07-02 by asimonov