PX4/PX4-Autopilot #9815: RM3100 Mag Driver - WIP

ryanjAA, 2018-07-01:
This is in a real hack job state. But after taking a brief pass it's building on master and reading some data form the sensor (single measurement only and far from working correctly but those better than me can probably shed some light). Anyway, this could be a promising mag replacement. Those I've spoken to that are using is have said its results have exceeded their expectations but no driver is available for px4 so here we start..

Datasheet/Manual: https://www.pnicorp.com/download/rm3100-user-manual/?wpdmdl=5678&ind=xr8oWECu6CPvLwLRp1LeLmWWCGnDLVNUCeQaWuH6h1A38i-f3XII6pzTdV_eamnboSAHCgi8cEHej-AyfacydQ

Sorry for the attachment below, not at computer for a bit. Anyone, want to add anything would be great.

[rm3100.cpp.zip](https://github.com/PX4/Firmware/files/2152405/rm3100.cpp.zip)

@dagar

dagar, 2018-07-01:
Let's get this in a pull request and start iterating.

ryanjAA, 2018-07-01:
ok - added see #9818

closed 2019-02-04 by stale[bot]