PX4/PX4-Autopilot #9824: bebop 2: cannot see drone in qgroundcontrol

asimonov, 2018-07-02:
**Describe the bug**
Cannot see Parrot Bebop 2 running px4 in qgroundcontrol

**To Reproduce**
Steps to reproduce the behavior:
1. built v1.8.0
2. uploaded to the drone
3. running px4.stripped on the drone. no issues in the on-screen messages
4. connected to Parrot wifi network on Ubuntu
5. running QGroundControl on Ubuntu shows 'waiting for vehicle connection'

**Expected behavior**
the connection established automatically

**Log Files and Screenshots**
# px4.stripped /home/root/px4.config 
commands file: /home/root/px4.config

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [param] selected parameter default file /home/root/parameters
130631 IMU temperature initialized to: 49.265293
166472 Initialize device: /sys/bus/iio/devices/iio:device0
169521 Initialize device: /sys/class/pwm/pwm_9
169974 Initialize device: /sys/class/gpio/gpio129
INFO  [bebop_flow] Set image format: 320x240
    Format: NV12
    Size: 115200
INFO  [bebop_flow] Streaming started: /dev/video0
WARN  [commander] LED: getHandle fail

WARN  [commander] LED init failed
INFO  [dataman] Unknown restart, data manager file '/data/ftp/internal_000/fs/microsd/dataman' size is 11405132 bytes
Sleeping for 1 s; (1000000 us).
INFO  [ekf2] Found range finder with instance 0
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
Sleeping for 1 s; (1000000 us).
INFO  [df_bebop_bus_wrapper] Trying to initialize mixers from config file /home/root/bebop.main.mix
INFO  [logger] logger started (mode=all)
pxh> INFO  [logger] Start file log
INFO  [logger] Opened log file: /data/ftp/internal_000/fs/microsd/log/sess002/log001.ulg
INFO  [mavlink] using network interface eth0, IP: 192.168.42.1
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 192.168.42.255

bramsvs, 2018-07-02:
This has always worked quite well for me. The PX4 messages are quite similar (same ports/IPs/data rates) so I suppose the problem sits at the ground side.. Are you using QGroundControl v3.3.2?

I get this message in PX4: `INFO  [mavlink] partner IP: 192.168.42.82`.

To which network are you connected exactly in step 4, `Bebop2-xxxxxx`? How do you upload/start PX4 when you aren't connected to the Bebop WI-FI network yet?

asimonov, 2018-07-02:
oops, was a firewall issue on the ground control PC

closed 2018-07-02 by asimonov