PX4/PX4-Autopilot #9829: Parameter missing from firmware

arvind6020, 2018-07-03:
For multirotor firmware roll & Pitch sensitivity slides to left .
it doesn't get adjusted after tuning .
it shows parameter missing though parameter isn't missing .

For all other configuration of vehicle it works fine.

arvind6020, 2018-07-03:
it doesn't solve the problem .
1 multirotor generic x config
[image]

LorenzMeier, 2018-07-03:
Please download the build from here:
https://docs.qgroundcontrol.com/en/releases/daily_builds.html

arvind6020, 2018-07-03:
Roll & Pitch Sensitivity parameters seems missing for multirotor frames , for Plane , Vtol all parameters are visible . 
[image]