PX4/PX4-Autopilot #9839: NSH Terminal throwing garbled and unreadable data

Ankur1014, 2018-07-05:
**Describe the bug**
Upon opening the NSH (Nuttx) Terminal, I am seeing junk and unreadable data

**To Reproduce**
Steps to reproduce the behavior:
1. Pixhawk board: Pixhawk 1
2. config: px4fmu-v2_default
3. Firmware and gcc: 1.8.0 and gcc 7.2
4. Board is connected over USB to host. 
5. Host Serial port: /dev/ttyACM0

**Expected behavior**
I want to login to the nuttx Terminal and run commands on the terminal. I have connected the board over USB to my Debian 9 system. I use the following command to open the terminal

screen /dev/ttyACM0 57600 8N1.

Upon doing so I am always getting this garbled screen.

[image]

I have restarted the host , reconnected the board but I always see same issue.

**Additional context**
I was working earlier with v1.6.5 but I did **not** face this issue of data being garbled at that time.

LorenzMeier, 2018-07-05:
You're seeing the MAVLink stream. Use the terminal function in QGC.

closed 2018-07-05 by LorenzMeier

Ankur1014, 2018-07-05:
@LorenzMeier ,
Does this mean that we cannot open the serial port /dev/ttyACM0 to get the data in Nuttx terminal anymore?