PX4/PX4-Autopilot #9841: Streaming identical messages

superware, 2018-07-05:
**Describe the bug**
Some messages are not streamed if nothing has changed in their data.

**To Reproduce**
Steps to reproduce the behavior:
1. Monitor the sending of POSITION_TARGET_GLOBAL_INT, see only a single packet per change.

**Expected behavior**
Most scenarios use unreliable wireless communication, it's unacceptable to expect a single packet to always arrive to the GCS.

See this [commit](https://github.com/PX4/Firmware/commit/3db287ba67917efd69d550a9240401f94587a845#commitcomment-29557774), @dagar

dagar, 2018-07-13:
In most cases this is a good mechanism for limiting irrelevant data from being published over a limited data link. We don't want mavlink publishing POSITION_TARGET_GLOBAL_INT if you're not in a mode that has one.

Here we could combine it with another publication to get the behaviour you want.

closed 2018-07-13 by LorenzMeier

superware, 2018-07-17:
@dagar,

1) "a good mechanism for limiting irrelevant data from being published over a limited data link", but still, that limited data link MUST have enough bandwidth to continuously transport ALL streams at their defined rates. How will limiting a low-rate stream message change that?
2) These limited data links are often wireless and unreliable, it might be catastrophic if the GCS will not know the current global targets just because the link failed to transport a single packet??

Thanks.

dagar, 2018-07-17:
1. There definitely isn't enough bandwidth to continuously transport ALL streams at their defined rates in most cases. Check `mavlink status`. On most serial links the transmit rates are automatically scaled back. 

2. I would reevaluate areas of the mavlink spec where you consider missing messages to be catastrophic. We have acknowledgements and retry logic that could be applied.

PR #9903 (2018-07-13): mavlink update POSITION_TARGET_GLOBAL_INT publish
 - publish continuously in position control mode when there's a valid
   setpoint
 - optionally fill in the velocity and acceleration setpoints when
   available
 - fixes #9841
commit 0c0b761c (2018-07-13): mavlink update POSITION_TARGET_GLOBAL_INT publish

 - publish continuously in position control mode when there's a valid
   setpoint
 - optionally fill in the velocity and acceleration setpoints when
   available
 - fixes #9841