PX4/PX4-Autopilot #9844: sf0x driver does not seem working.

weiminshen99, 2018-07-06:
@LorenzMeier

I am following up with Issue #4635 and testing a new sensor SF10/a by connecting it to SERIAL/4, and set SENS_EN_SF0X = 2. It does not seem working properly and nsh shows that all readings from the sensor are errors (see below). Could you please tell me what I did wrong? FYI, I also tested the TFMini sensor on SERIAL/4  and it works nicely. Thanks for your help!

===================

> python mavlink_shell.py
Connecting to MAVLINK...

nsh> sf0x info
state @ 1000b0e0
sf0x_read: 10908 events, 94312us elapsed, 8us avg, min 6us max 318us 14.305us rms
sf0x_com_err: 10908 events
poll interval: 31 ticks
report queue	2/64 (2/2 @ 1000b1a0)

nsh> sf0x stop

nsh> sf0x start

nsh> sf0x info
state @ 1000b0e0
sf0x_read: 230 events, 1879us elapsed, 8us avg, min 6us max 111us 9.146us rms
sf0x_com_err: 231 events
poll interval: 31 ticks
report queue	2/64 (2/2 @ 20025690)
nsh>

xdwgood, 2018-07-06:
1

weiminshen99, 2018-07-09:
I found the reason why it is not working. The factory default buadrate of SF10/a is 115200, but PX4 SF0X driver's expected baudrate is 9600. To change the baudrate of SF10/a, you can use the product's USB cable connecting to a PC running their terminal program. When you see the values are displayed, you can type "space" and then "c" to change the baudrate.

I hope this information will be useful for future users with similar problems.

LorenzMeier, 2018-07-10:
@DanielePettenuzzo Is this a general issue?

DanielePettenuzzo, 2018-07-10:
@LorenzMeier seems that all the range finders supported by this driver can be configured with a baud between 9600 and 115200. For example SF02 has a default baud equal to 9600 while the SF10/a has 115200. I was not able to find the factory default for each different sensor because datasheets just mention the possible range.. should we look for all these default baud rates and set them for each one in the driver?

LorenzMeier, 2018-07-10:
Let's do the two that we know about.

closed 2019-02-04 by stale[bot]