PX4/PX4-Autopilot #9845: PX4 ver 1.8 control lost & unexpected behaviour

albert21martins, 2018-07-06:
Hi,

By performing a survey, the copter went unstable at aprox. minute 6, we lost control, but thanks to geofance  could be saved. After reviewing the log, we cannot say what went wrong. Our internal logs show that the motor 8 started abnormal behaviour at min 4 and broken at min 5. 
[image]

Log here:
https://logs.px4.io/plot_app?log=dd3c8aa9-9d77-428d-8054-ce055f5ec0ef

The system is:
- Octocopter coax, up to 5 Kg
- Pixhawk 2.1, ver 1.8
- Here GPS & 3DR GPS for magnetometer and secondary GPS, as there is nor driver implemented yet for HERE GPS. Btw, the 3DR GPS should be set 180º magnetometer orientation in order to work (bug here?).

Does anybody has any idea what could be the source of our problem?

Sincerely,

Albert

LorenzMeier, 2018-07-06:
Looks like a mechanical or electrical problem. The difference in outputs is a tell-tale sign.

kuiskas, 2018-07-06:
Hi Lorenz,

thanks for the fast answer. We also have a custom board as Power supply 1, and the standard from the Pixhawk 2 cube as a secondary Supply. I realized the Power supply 1 is not working anymore. Could this somehow perform a brownout in the system and perform a reboot? What happens if the system restart while flying?

LorenzMeier, 2018-07-06:
You can check in the flight log with a tool called Flight Plot. But its not the root cause, because the log file continued and you clearly have an issue with your actuators.

I'm closing this because its not an issue with the upstream software but with your specific airframe. Feel free to post on http://discuss.px4.io for community support or alternatively contact one of the consulting companies which offer commercial support.

closed 2018-07-06 by LorenzMeier