PX4/PX4-Autopilot #9848: Cant upload waypoints to Intel Aero using QGC

ambikaverma, 2018-07-06:
Hi
I have seen others also face this issue -

When try to add waypoints for navigation the Qgroundcontrol is not able to send the waypoint to the intel aero and continuously fails due to Operation timeout error. Even if I send a single waypoint.

AeroFC firmware version - 1.6.5
QGC version - 3.3
Other solutions have been to switch to ardupilot firmware, but this should work with px4 also.
Any fixes for this issue?
Thanks!

r0gelion, 2018-07-06:
Hi!

First, I would update the operative system, BIOS and FPGA, here is a video on how to do it: https://youtu.be/e9MLnRbMDHo

Then, update the PX4 firmware to the current stable version which is 1.8.0, download the aerofc-v1_default.px4 file from here: https://github.com/PX4/Firmware/releases

You need to copy the px4 file to the Intel Aero computer in "/etc/aerofc/px4/". If you are using windows you can use WinSCP to transfer the file.

Once the file is there you need to connect to the Intel Aero by SSH and execute the commands:
```
cd /etc/aerofc/px4/
aerofc-update.sh aerofc-v1_default.px4
```

More information at https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup

I hope this helps!

ambikaverma, 2018-07-14:
Hi

Thanks. doing the version update did help fix the issue!

closed 2018-07-14 by ambikaverma