PX4/PX4-Autopilot #9859: Missing PWM_MAIN_REV* params in v1.8.0

IeiuniumLux, 2018-07-08:
I'm not longer able to see the PWM_MAIN_REV* params in v1.8.0 release; neither in QGC v3.3.2 or in the _nsh_ console.  The ailerons of my Wing Wing Z-841 are moving in the opposite of the correct stabilization direction. If these params have been removed in  v1.8.0 release, then which param should I now use to revert the roll correction in stabilized mode?

[image]
[image]

LorenzMeier, 2018-07-16:
Which flight controller are you using?

IeiuniumLux, 2018-07-16:
[mRo PixRacer R14](https://store.mrobotics.io/mRo-PixRacer-R14-Official-p/auav-pxrcr-r14-mr.htm)

LorenzMeier, 2018-07-16:
[image]

With a current Pixhawk 4 it all looks good. I'm closing this for now.

IeiuniumLux, 2018-07-16:
Ok, but I'm not using the Pixhawk 4.  Did you test it with the Pixracer?

LorenzMeier, 2018-07-16:
@hamishwillee Can we document this in the user guide? We should change the comment string so it discusses the autopilot component MAIN / AUX are each connected to.

The naming is unfortunate:

- MAIN: IO
- AUX: FMU

Pixracer has only FMU, so it has only AUX. I need to figure out why the *_DIS parameters show up at all - there should be none there.

LorenzMeier, 2018-07-16:
Fix in code is here:
https://github.com/PX4/Firmware/pull/9917

@hamishwillee Please close the issue once documentation is updated.