PX4/PX4-Autopilot #9863: Bug when building

L1ttlewhite, 2018-07-09:


LorenzMeier, 2018-07-09:
Your GCC version is too outdated. Please update your toolchain.

mlanting, 2018-08-09:
I had this issue too, but had to copy the header mentioned by @ProfFan to the `src/include` directory as well to get it to build.

jonathan84clark, 2018-08-09:
This worked for me too. For those of you who do not have the entire directory structure of PX4 memorized the location I placed the file in was '/Firmware/platforms/nuttx/NuttX/nuttx/include'
cd into that directory then do a 
wget https://raw.githubusercontent.com/esp8266/Arduino/master/tools/sdk/libc/xtensa-lx106-elf/include/_ansi.h 
Then 'make clean'
Then try your build.

mlanting, 2018-08-10:
Should this be re-opened or filed as a new issue, since it seems to be more than an out-of-date toolchain problem?

dagar, 2018-08-10:
I'd still strongly recommend everyone sync up with GCC 7 specified in the dev guide. http://dev.px4.io/en/setup/dev_env.html
The project is only going to officially support a single version at a time.

That being said, I'm not opposed to tracking down the root cause here. Copying a random libc header over definitely isn't the right solution.

@ProfFan how did you come up with that? Could you or @mlanting post the full verbose build output (formatted)? I would guess this is far more likely to be a problem with include paths.

mlanting, 2018-08-10:
I wasn't thinking about this earlier, but I was following the [build instructions for the Intel-Aero drone](https://docs.px4.io/en/flight_controller/intel_aero.html) specifically, which asks the user to run `make aerofc-v1_default` rather than just `make`. When I first tried to recreate the output with a fresh checkout this morning just using `make` it worked fine.

I've attached the full output of `make aerofc-v1_default VERBOSE=1`
[firmware_build_fail_output.txt](https://github.com/PX4/Firmware/files/2278497/firmware_build_fail_output.txt)

dagar, 2018-08-10:
Oh, I bet it's just a race condition. A certain amount of the nuttx build has to be configured and copied into place before other parts of firmware can be built.

dagar, 2018-08-10:
@mlanting there's a dependency missing somewhere for DriverFramework. In the meantime could you try with the cmake ninja-build generator? Simply install ninja or ninja-build, clean, rebuild.

@L1ttlewhite please post your full build output.

mlanting, 2018-08-10:
I found a way to run ninja on just that target and I'm getting similar errors:
[firmware_build_fail_ninja.txt](https://github.com/PX4/Firmware/files/2278896/firmware_build_fail_ninja.txt)

dagar, 2018-08-10:
@mlanting what's your exact environment?

ProfFan, 2018-08-12:
Just FYI:
After copying _ansi.h manually:
```
$ find . -name '_ansi.h'
./build/nuttx_px4fmu-v4_default/NuttX/nuttx/include/_ansi.h
./platforms/nuttx/NuttX/nuttx/include/_ansi.h
```
Which suggests that NuttX does not have built-in _ansi.h
```
$ ag _ansi.h
platforms/nuttx/NuttX/nuttx/arch/arm/include/math.h:
6:#include "_ansi.h"
```
However PX4 NuttX `math.h` requires `_ansi.h`.

Meanwhile, in freshly-cloned official NuttX:
```
$ find . -name 'math.h'
./nuttx/include/nuttx/lib/math.h
```
I think there must be some divergence between the official NuttX and the PX4 fork.

A simple `git blame` shows that `platforms/nuttx/NuttX/nuttx/arch/arm/include/math.h` is added by @dagar and is the only occurrence of `_ansi.h` usage.

mlanting, 2018-08-13:
I am running Ubuntu 16.04 on an intel-aero RTF drone set up according to these instructions:
https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation

When I got to the [Flight Controller](https://github.com/intel-aero/meta-intel-aero/wiki/90-(References)-OS-user-Installation#flight-controller) step, I wanted a fresh build of the firmware from the latest in the Firmware repository, so I followed the [for more details](https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-Setup#flashing-the-flight-controller-rtf-only) link which pointed me to a page in the [PX4 wiki](https://docs.px4.io/en/flight_controller/intel_aero.html). I followed the instructions there starting under the heading "Flashing PX4 software" (which I can't seem to direct link). Step 3 in that section is where I encountered the issues that led me here.

That's about as exact as I can describe it - let me know if anything is unclear.

dinesh286, 2018-08-21:
### 
I am getting the following error during compilation
i have upgraded gcc-arm and included _ansi.h in the proper folder

 sudo make px4fmu-v3_default
-- CMAKE_MODULE_PATH: /mnt/c/PX4/Firmware/cmake
-- Nuttx build for px4fmu-v3 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- Build Type: MinSizeRel
-- PX4 VERSION: PixhawkPSP_v3.0.3
-- CONFIG: nuttx-px4fmu-v3-default
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12")
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2
-- C compiler: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
-- Using C++03
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- Adding ROMFS on px4fmu-v3
-- Nuttx build for px4fmu-v3 on m3 hardware, using nsh
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/PX4/Firmware/build_px4fmu-v3_default
[74/767] Building NuttX for px4io-v2 with nsh
FAILED: cd /mnt/c/PX4/Firmware/build_px4fmu-v3_default/px4io-v2 && /bin/rm -rf /mnt/c/PX4/Firmware/build_px4fmu-v3_default/px4io-v2/NuttX/nuttx/nuttx-export && /usr/bin/make --quiet --no-print-directory -C /mnt/c/PX4/Firmware/build_px4fmu-v3_default/px4io-v2/NuttX/nuttx -r CONFIG_ARCH_BOARD=px4io-v2 export >nuttx_build.log && cmake -E touch /mnt/c/PX4/Firmware/build_px4fmu-v3_default/px4io-v2/nuttx_export.stamp
stdio/lib_sscanf.c: In function 'vsscanf':
stdio/lib_sscanf.c:419:28: error: this statement may fall through [-Werror=implicit-fallthrough=]
                       sign = true;
                            ^
stdio/lib_sscanf.c:420:21: note: here
                     case 'u':
                     ^~~~
cc1: all warnings being treated as errors
make[2]: *** [bin/lib_sscanf.o] Error 1
make[1]: *** [libc/libc.a] Error 2

jonathan84clark, 2018-08-21:
There are a few things you can try. 
1. Make sure all dependencies are installed by sourcing ubuntu_sim_nuttx.sh https://dev.px4.io/en/setup/dev_env_linux.html
2. Do a 'git submodule update --init --recursive
3. Do a 'make clean' or/and remove your build directory
I ran into similar issues to these when I first started out. Unfortunately there are multiple sets of instructions for building PX4. No single set of instructions can be followed to have a successful build, usually you need to go through each set of instructions to fill in the missing pieces.
https://dev.px4.io/en/setup/dev_env_linux.html 
https://dev.px4.io/en/setup/building_px4.html
https://github.com/PX4/Firmware/blob/master/README.md
And now there are a number of smaller issues that were created when updates were made to dependent repositories.
For example, now in order to build you need to (in addition to following the above instructions) do the following (from my post earlier):

> This worked for me too. For those of you who do not have the entire directory structure of PX4 memorized the location I placed the file in was '/Firmware/platforms/nuttx/NuttX/nuttx/include'
cd into that directory then do a
wget https://raw.githubusercontent.com/esp8266/Arduino/master/tools/sdk/libc/xtensa-lx106-elf/include/_ansi.h
Then 'make clean'
Then try your build.

It would be nice to have a concise set of instructions that explain how to get everything setup on Linux that doesn't have a bunch of links to dependencies and workarounds.

dinesh286, 2018-08-22:
I followed the instructions for installing all the packages.  but i am getting same error.  the problem is it is showing warnings in the file fs_lseek.c, math.h, stdio/lib_sscanf.c, lib_dtoa.c a typical warning is 
stdlib/lib_strtold.c:178:9: note: here
         case '+':
         ^~~~
cc1: all warnings being treated as errors

Since warning are treated as error, it is not creating .o file

help me how to go abt

siddhu45, 2018-12-05:
Hi Guys, I know this is old but I am trying to build PX4 v1.6.5 on my Ubuntu 16.04 and it is failing. I can run the gazebo simulation from running `make posix_sitl_default gazebo` but `make px4fmu-v2_default` and `make px4fmu-v2_lpe` fails with the following message:

> sid@sid-Desktop:~/src/Firmware$ make px4fmu-v2_lpe 
> -- CMAKE_MODULE_PATH: /home/sid/src/Firmware/cmake
> -- Nuttx build for px4fmu-v2 on m4 hardware, using nsh with ROMFS on px4fmu_common
> -- Build Type: MinSizeRel
> -- PX4 VERSION: v1.6.5
> -- CONFIG: nuttx-px4fmu-v2-lpe
> -- The ASM compiler identification is GNU
> -- Found assembler: /usr/bin/arm-none-eabi-gcc
> -- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
> -- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
> -- C compiler: arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
> -- C++ compiler: arm-none-eabi-g++ (15:5.4.1+svn241155-1) 5.4.1 20160919
> -- Adding ROMFS on px4fmu-v2
> -- Nuttx build for px4fmu-v2 on m3 hardware, using nsh 
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/sid/src/Firmware/build_px4fmu-v2_lpe
> [73/584] Generating ../../../../parameters.xml
> OVERRIDING SYS_MC_EST_GROUP to 1!!!!!
> [73/584] Building NuttX for px4fmu-v2 with nsh
> FAILED: cd /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2 && /bin/rm -rf /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/NuttX/nuttx/nuttx-export && /usr/bin/make --quiet --no-print-directory -C /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/NuttX/nuttx -r CONFIG_ARCH_BOARD=px4fmu-v2 export >nuttx_build.log && cmake -E touch /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/nuttx_export.stamp
> In file included from /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/NuttX/nuttx/include/math.h:52:0,
>                  from ./stdio/lib_libdtoa.c:47,
>                  from stdio/lib_libvsprintf.c:236:
> /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/NuttX/nuttx/include/arch/math.h:5:28: fatal error: machine/ieeefp.h: No such file or directory
> compilation terminated.
> ERROR: arm-none-eabi-gcc failed: 1
>        command: arm-none-eabi-gcc -MT bin/lib_libvsprintf.o  -M -std=gnu99 -Wall -Wextra -Werror -Wdouble-promotion -Wframe-larger-than=1024 -Wlogical-op -Wpacked -Wpointer-arith -Wshadow -Wno-sign-compare -Wno-unused-parameter -Wno-nonnull-compare -Wno-misleading-indentation -Wbad-function-cast -Wmissing-parameter-type -Wnested-externs -Wstrict-prototypes -Wno-bad-function-cast -Wno-cpp -Wno-implicit-function-declaration -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-nested-externs -Wno-old-style-declaration -Wno-pointer-sign -Wno-type-limits -Os -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer -funsafe-math-optimizations -fno-builtin-printf -ffunction-sections -fdata-sections -g -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -I. -isystem /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/NuttX/nuttx/include -D__KERNEL__ -pipe -fno-common stdio/lib_libvsprintf.c
> make[2]: *** [.depend] Error 1
> In file included from /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/NuttX/nuttx/include/math.h:52:0,
>                  from ./stdio/lib_libdtoa.c:47,
>                  from stdio/lib_libvsprintf.c:236:
> /home/sid/src/Firmware/build_px4fmu-v2_lpe/px4fmu-v2/NuttX/nuttx/include/arch/math.h:5:28: fatal error: machine/ieeefp.h: No such file or directory
> compilation terminated.
> make[2]: *** [bin/lib_libvsprintf.o] Error 1
> make[1]: *** [libc/libc.a] Error 2
> [73/584] Building NuttX for px4io-v2 with nsh
> ninja: build stopped: subcommand failed.
> Makefile:164: recipe for target 'px4fmu-v2_lpe' failed
> make: *** [px4fmu-v2_lpe] Error 1

I have tried with older version 4.9.3 of `arm-none-eabi-gcc` as well and it too doesn't work. I have to modify the something in particular in the code and rebuild the firmware. Any ideas what is going wrong. This is my regular PC and have all the latest updates (apt updates) are up-to-date.
I have an old PC with Ubuntu 16.04 which is not updated and I can build the PX4 v1.6.5 there. The compiler version `arm-none-eabi-gcc` is 4.9.3 on that PC. I tried to read all the issues others are facing and everyone suggests to update the gcc to a newer version but still how can I make this build.
Thanks for the help!

closed 2019-03-19 by stale[bot]

fpvaspassion, 2019-04-22:
Hello,
    Whe i compile latest master - this issue does not appear, when i checkout to v1.8.2 - i have this issue and only wget of _ansi.h header fle helps to compile code.  Gcc version is 7.3.1