PX4/PX4-Autopilot #9866: NaN rate setpoints when switching from acro to manual

bkueng, 2018-07-09:
The log https://logs.px4.io/plot_app?log=89ab485c-7768-4fca-b7e9-5ed9d09152b3 contains NaN setpoints when switching from acro to manual. Seems to be a race condition when the position controller did not yet publish a valid attitude setpoint but the attitude controller already got the mode switch.

However the position controller does publish an invalid setpoint with constant timestamp before that, as seen in the CSV file:
```
timestamp,roll_body,pitch_body,yaw_body,yaw_sp_move_rate,q_d[0],q_d[1],q_d[2],q_d[3],thrust,landing_gear,q_d_valid,roll_reset_integral,pitch_reset_integral,yaw_reset_integral,fw_control_yaw,disable_mc_yaw_control,apply_flaps
[...]
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
10039547,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,-1.0,0,0,0,0,0,0,0
234508568,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.118343,-1.0,1,0,0,0,0,0,0
234529628,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.118343,-1.0,1,0,0,0,0,0,0
234550563,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.118343,-1.0,1,0,0,0,0,0,0
234571562,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.120316,-1.0,1,0,0,0,0,0,0
234592561,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.123274,-1.0,1,0,0,0,0,0,0
234613560,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.12426,-1.0,1,0,0,0,0,0,0
234634566,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.12426,-1.0,1,0,0,0,0,0,0
234655562,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.126233,-1.0,1,0,0,0,0,0,0
234676562,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.12426,-1.0,1,0,0,0,0,0,0
234697562,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.125247,-1.0,1,0,0,0,0,0,0
234718560,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.12426,-1.0,1,0,0,0,0,0,0
234739571,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.125247,-1.0,1,0,0,0,0,0,0
234760563,0.0,-0.0,1.90531,0.0,0.579523,0.0,0.0,0.814956,0.126233,-1.0,1,0,0,0,0,0,0
234781563,0.0,0.167662,1.90531,0.0,0.577487,-0.0682384,0.0485249,0.812094,0.127219,-1.0,1,0,0,0,0,0,0
234802565,0.0,0.206141,1.90531,0.0,0.576447,-0.0838494,0.059626,0.810631,0.127219,-1.0,1,0,0,0,0,0,0
234823563,0.0,0.241872,1.90531,0.0,0.57529,-0.0983176,0.0699145,0.809004,0.127219,-1.0,1,0,0,0,0,0,0
```

[log001_vehicle_attitude_setpoint_0.zip](https://github.com/PX4/Firmware/files/2175558/log001_vehicle_attitude_setpoint_0.zip)

@MaEtUgR @Stifael can you look into this please?

dagar, 2018-07-13:
It might be due to MPC operating without a position.

MaEtUgR, 2018-07-24:
Based on what commit were you flying? The hash is not known on master. I'm asking because this might have changed with #9731... or was it already including these changes?

It's hopefully reproducible in simulation.
> It might be due to MPC operating without a position.

Is a good idea.

bkueng, 2018-07-24:
> I'm asking because this might have changed with #9731

No it was before that. I was wondering that as well, it could be fixed already. I can report back when I see it again.

MaEtUgR, 2018-09-20:
@bkueng Does this persist or can we close?