PX4/PX4-Autopilot #9869: Feature Request: Failsafe to non-airspeed TECS mode when airspeed data goes bad or missing

Antiheavy, 2018-07-10:
**Problem:**

Airspeed measurement with a Pitot-static probe and differential pressure sensor is prone to problems in real world situations.  For example: pitot tubes can easily become plugged with water or dirt and pneumatic tubes can easily develop leaks.  Very bad airspeed measurements can lead to very bad flight performance and crashes.

**Proposed solution:**

Develop an airspeed failsafe system in Commander to detect bad airspeed measurements using EKF2's estimator_status.tas_test_ratio to determine when airspeed measurements are bad.

To avoid false positives:
-  Threshold values for flagging estimator_status.tas_test_ratio both bad and good need to be found.  E.g. a test ratio >2 may be required to flag as "bad" and then the test ratio may need to fall back <1 to call "good" again.   Actual values TBD, probably should be parameters.
- A time duration (hysterisis) will be required for determining when to switch back and forth between airspeed and non-airspeed modes based on the test ratios. value TBD  hard code or new parameter(s)?
- Add the condition that mag_test_ratio and vel_test_ratio must also be < 1.

At least one new parameter is needed: COM_ASPD_FS_ACT

Failsafe action when bad airspeed measurements are detected (default 0):
0 log a message
1 log a message, warn the user
2 log a message, warn the user, switch to non-airspeed TECS mode
3 log a message, warn the user, switch to non-airspeed TECS mode, switch to Return mode

If a complete loss of airspeed data occurs during flight (e.g. an electrical fault) the system should automatically perform failsafe option 2 unless 3 is selected.  There may be some existing behavior here, maybe it should be rolled into the failsafe actions?

A related, but not dependent, feature request is an optional pre-flight functional test for airspeed that can be called from the QGC Pre-flight checklist (see separate PX4 and QGC feature requests)

Antiheavy, 2018-07-10:
FYI @dagar @priseborough @kjkinney @m-skelton

dagar, 2018-07-10:
This airspeed architecture issue is relevant. https://github.com/PX4/Firmware/issues/9148

dagar, 2018-07-10:
Notes from a previous conversation with @priseborough about using EKF2 airspeed innovation fault status as part of an airspeed validity check.
 - we could provided a persistence criteria was applied to prevent wake crossing and other gust events causing false positives
 - A known relationship between pitch, density altitude and throttle can also be used to improve detection accuracy.

Antiheavy, 2018-07-23:
From a recent conversation with @priseborough : 

"We can implement a two stage process. The  test ratios would be used to detect the presence of a airspeed error which arms to arm the airspeed mode change and alerts the operator. Once that has occurred the reversion would happen immediately that loss of height control is detected. "

Antiheavy, 2018-08-09:
Often an airspeed measurement fault will manifest during takeoff (e.g. plugged or leaky pitot-static) so the failsafe might need to happen quickly to avoid crashing in that case.  Maybe don't wait for the loss of altitude during takeoff mode, but trip the failsafe immediately upon detection of an issue.  If we trip a false positive during takeoff mode, hopefully after altitude is reached the system would switch back to normal airspeed mode.

Antiheavy, 2018-09-15:
Here are some additional logs that may help towards this:

blocked pitot:
https://review.px4.io/plot_app?log=77d71e51-c131-45c1-92c2-aa5bf14ef66e

Leaky pitot:
https://review.px4.io/plot_app?log=c0ed0754-c063-402e-a0b2-e34b74bb2164

Bad compass calibration, shows airspeed checkbits too (maybe false?):
https://review.px4.io/plot_app?log=8073ef78-2f58-4e4e-91f9-471649667898

priseborough, 2018-09-19:
This will require an understanding of the limitations of the estimator, TECS and air data failure modes. The solution will likely require changes in multiple modules.

priseborough, 2018-09-19:
Replay logs are good to understand how the errors are affecting the estimator. However because detection will change flight vehicle response and therefore sensor data, developing a representation of these failure modes in SITL is also required and should be the first activity.

Antiheavy, 2018-10-09:
Here is another flight log that shows some airspeed innovations that could be used as false positive example (also, maybe there is a bug with what is happening on the ground vs in the air?):
https://review.px4.io/plot_app?log=7498be6c-0e18-4ea4-963b-c5c8c2456937

priseborough, 2018-10-11:
I've created a prototype here: https://github.com/priseborough/Firmware/tree/airSpdFailsafe-wip

The next steps are:

- Make the required SITL mods to test it.
- Evaluate use of height loss, throttle and current draw criteria as an additional criteria.

priseborough, 2018-10-15:
I was able to make some further progress on this and in addition to the EKF innovation checks, have added a normal load factor sanity check that flags when the measured air speed is not plausible given the measured aerodynamic load factor. This relies on the correct setting of the COM_ASPD_STALL parameter but does enable faster detection of pitot tube blockage after launch.

priseborough, 2018-10-15:
I evaluate another check based on pitch angle, throttle and airspeed, but it required more physical parameters to be set and was not as fast as the normal load factor method in detection of a blocked pitot tube condition so abandoned the approach.

We will also need to modify the takeoff controller to abort of the airspeed doesn't rise by the time ground speed has exceeded the stall speed. Hand/catapult/bungee launches are OK at the moment with the fault detection, however if the pitot tube is blocked during a ROG takeoff, the vehicle runs off the end of the runway and doesn't rotate and transition into a flying state. A simple takeoff distance check would suffice. If sufficient airspeed to rotate is not gained after the specified distance travelled, the engine should be cut and the takeoff should be aborted. This should be handled separately in the takeoff controller logic.

Antiheavy, 2019-01-20:
still an open feature request pending completion of https://github.com/PX4/Firmware/pull/10733 or similar.  thanks.

AuterionWrikeBot, 2019-04-14:
➤ Wrike Bot commented:

Paul Riseborough All subtasks have been completed or deferred, you might want to complete this task as well.

Antiheavy, 2019-07-13:
this works!  thanks.

closed 2019-07-13 by Antiheavy