PX4/PX4-Autopilot #9873: COMMAND_LONG Payload Cut-off

juniordiscart, 2018-07-10:
**Describe the bug**
In a SITL simulator environment, a COMMAND_LONG message is sent from the firmware to request the sendrate of HIL_STATE_QUATERNION messages. According to the COMMAND_LONG message definition in [common.xml](https://mavlink.io/en/messages/common.html#COMMAND_LONG), the payload should be 33 bytes. However, the received COMMAND_LONG message has a payload of just 30 bytes.

This causes some auto-generated MAVLink message parsing tools to break, i.e. C#, where a fixed amount of bytes are expected. This started from v1.8.0. The problem is not present in v1.7.3.

Note in the screenshot below the second byte (1e) which depicts the payload size, which translates to a value of 30.

[image]

mhkabir, 2018-07-10:
Mavlink v2 chops off zeros at the end of the payload. This is not a bug. You need to update your C# application to use the v2 parsers.

LorenzMeier, 2018-07-10:
Closing accordingly.

closed 2018-07-10 by LorenzMeier