PX4/PX4-Autopilot #9879: HIL or HITL: PWM signal not working

Dohyeung, 2018-07-11:
I am using XPlane10, QGC and firmware 1.5.5 for HIL testbed. It works well but I am trying to generate PWM signal to control actuators of Easystar II RC plane. I modified the 'state_machine_helper.h' file which is in the /firmware/scr/modules/commander folder. In the 186 line of the file, I changed 'armed->lockdown = true;' to 'armed->lockdown = false;'. Then, everything does not work now. Could you help me if you are an expert for this problem.

Best regards.

LorenzMeier, 2018-07-16:
Are you using firmware version 1.5.5?

Dohyeung, 2018-07-16:
@LorenzMeier Yes. I am using v1.5.5 to run the HIL. If I do not want to run motors of EasyStar II, everything is working in HIL. However, if I want to run its servo with the modified firmware, it does not work at all.

LorenzMeier, 2018-07-16:
Sorry, that version of the Firmware is fairly old. We only cover issues on recent versions (1.8).

closed 2018-07-16 by LorenzMeier

Dohyeung, 2018-07-16:
@LorenzMeier OK. I got it but I think I cannot use the versions (1.8) for HIL. Am I wrong? I saw several posts that I had to use v1.5.5 for HIL. Could you give me some advices?