PX4/PX4-Autopilot #9889: RC not working in HIL (HIL_ACTUATOR_CONTROLS message not send)

pascalgross, 2018-07-12:
**Describe the bug**
In HIL mode (X-Plane or Simulink), no surface deflections can be seen.

**To Reproduce**
- Current 'master' or 'stable'
- SYS_AUTOSTART = 2100 (Standard Plane)
- HITL Enabled
- Steps to reproduce the behavior:
- Connect to X-Plane
- (GPS_RAW_INT is fired)
- (Manual mode is active)
- Move the RC sticks around

**Expected behavior**
- I would expect the control surfaces to move.
- I would expect to see HIL_ACTUATOR_CONTROLS message in MAVLink Inspector.

**Additional context**
- ACTUATOR_CONTROL_TARGET is changing with stick inputs.

closed 2019-02-04 by stale[bot]

v01d, 2019-07-25:
Hello,
I have a similar problem when trying to connect rotors simulator with px4 running in HIL mode. Once I arm, I get a stream of zero hil actuator controls. In the past this was due to this missing module, but I can confirm it is running on my board. What could be happening?

possibly related: #11697

julianoes, 2019-07-26:
@v01d your issue sounds different, I'd open a new one and give some more context and details, or re-open #11697 and add your information there.