PX4/PX4-Autopilot #9890: Indoor localization with drone on px4flow and sonar only.

rahulsharma11, 2018-07-12:
Dear team,
I am working with pixhawk and v3 firmware on my single copter. I made this for indoor flight only (like a personal assistance). Till now i am able to make it fly and hold the position, thanks to the team for support to achive this. 
As i am intended for indoor localization only so GPS will not work but for any mission GPS is required.
I will suggest if there can be an option for the user to do any desired mission indoor or outdoor,  thus have an option to use GPS or totally rely on px4flow camera and sonar only for mission.
Is it possible to have that? Any alternative solution that can be used for the same?
Thanks.

TSC21, 2018-07-12:
From what I know, currently only LPE has the feature to allow missions to be processed without an available GPS. Thus, there will be further developments on EKF2 to allow px4flow only missions indoor/outdoor.

rahulsharma11, 2018-07-12:
But now LPE is not supported any more by the team i guess. Only EKF is supported.

TSC21, 2018-07-12:
LPE is not officially supported. But it doesn't mean you cannot use it ans ask for support about it.

rahulsharma11, 2018-07-12:
I had used lpe before and was not able to hold drone's position with lpe. I switched to ekf then and it gave me good result. Therefore i myself not wanted to go back in lpe. BTW if lpe is only way to do indoor localization. I will give another try to that till any update is there in EKF.

TSC21, 2018-07-12:
Is not the only way to do localization indoor - is the only way to allow doing missions without needing a GPS.

rahulsharma11, 2018-07-12:
Ok. Thanks for clarification.

rahulsharma11, 2018-07-12:
Ok one last question, if i am using offboard mode. Raspberry with pixhawk, i will provide the waypoints through mavlink communication. So does that also require GPS for mission?

Oceasdeep, 2018-07-22:
@rahulsharma11 .now i have a problem ,with the 1.8.0 firmware my px4flow didn't working ,there has no localposition infomation.what can i do?waiting your reply,thanks advanced.

rahulsharma11, 2018-07-23:
Hi @Oceasdeep , Firstly check your optical flow is been determined after pixhawk boot or not. You can see that in Mavlink_Inspecter option in widgets tab in qgroundcontrol. Then please adjust lense focus  by connecting it with qgroundcontrol and check live plot of OPTICAL_FLOW_RAD.quality parameter. It should be above 200.

adipandas, 2018-09-12:
@rahulsharma11 Reply for your previous question
> Ok one last question, if i am using offboard mode. Raspberry with pixhawk, i will provide the waypoints through mavlink communication. So does that also require GPS for mission?

So, if you are doing offboard indoors, you do not have to use GPS for mission. You can disable the GPS by unchecking GPS and enabling optical flow option in EKF2_AID_MASK parameter using QGroundControl.
Hope this helps.

adipandas, 2018-09-12:
@rahulsharma11 I have few questions regarding the use of optical flow though.
I am also facing some issues with px4Flow. It will great if you can help me with these.
I am trying to get my quadrotor to hover indoors without GPS/MOCAP. I have firmware version 1.8.0 on my pixhawk. I am using EKF2 to work with Optical Flow.

Can you please let me know how did you get it to hover indoors.
Following are the things I am doing while working with pixhawk to just get my quadcopter to hover indoors:
1. Use opticalflow - PX4Flow.
2. I checked the parameters as per you suggested in [here](https://github.com/PX4/Firmware/issues/9890#issuecomment-406950964)
3. I enabled option for "optical flow" sensor in the parameter `EKF2_AID_MASK` and disabled GPS.

But before I try to fly in `offboard` mode, I want to make sure the platform is able to hold the position indoors using only optical flow feedback. Can you please let me know what steps did you take to test the position hold indoors?

Thanks in advance.
Waiting for your reply.

tingtingxing, 2018-10-31:
> @rahulsharma11 .now i have a problem ,with the 1.8.0 firmware my px4flow didn't working ,there has no localposition infomation.what can i do?waiting your reply,thanks advanced.

I am facing the same issue with you. I have kept the quality over 200 and changed the parameter EKF2_AID_MASK to 2. But that didn't work. Did you solve the problem "no localposition information"? If yes, could you please tell me what can I do for that?

rahulsharma11, 2018-10-31:
Hi @adipandas ,
After verifying the checks, you can directly start with Auto_takeoff. You should provide the max altitude upto 1/2 meters and start with Autotakeoff. And then do hold after takeoff. Thus you can verify the position hold is working or not. Make sure you are getting flow data above 200 values at calibrated height.

tingtingxing, 2018-10-31:
Hellow @rahulsharma11 
Thanks so much for your reply. I have checked the optical quality and Sonar distance in QGC, the result is showing in Mavlink Inspector below:
[image]

Are these two data correct?

I am just starting working on PX4 so I don't know how to use logs function yet. I am going to check my logs as soon as I find the right place to check them.

rahulsharma11, 2018-10-31:
Are you using separate I2C sonar? better to use that instead of sonar that is embedded with px4flow.
If yes, then you are good to go.

closed 2019-02-13 by stale[bot]