PX4/PX4-Autopilot #9900: Lsm303d and l3gd20 are not recognized on pixhack v3

cuhome, 2018-07-13:
FMUv2 ver 0xA : Rev 0 V2M
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4FMU_V2
HW type: V2M
HW version: 0x0009000A
HW revision: 0x00000000
FW git-hash: 96443b3cf3e3adaec8b2ebb6c83a19a92d40a13f
FW version: Release 1.8.0 (17301759)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 63775322bf25adb406594f8e610122fe0cef2f7a
Build datetime: Jun 19 2018 06:39:00
Build uri: BUILD_URI
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
MFGUID: 33343537313751020052003b
MCU: STM32F42x, rev. 3
UID: 52003B:31375102:33343537 
[hardfault_log] Fault Log is Armed
INFO  [tune_control] Publishing standard tune 1
INFO  [param] selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
nsh: rgbled_pwm: command not found
MS5611_SPI on SPI bus 4 at 3 (20000 KHz)
WARN  [ms5611] no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 2 (type: 1)
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 4 at 4 (1000 KHz)
INFO  [mpu6000] accel cutoff set to 30.00 Hz
INFO  [mpu6000] gyro cutoff set to 80.00 Hz
ERROR [l3gd20] driver start failed
WARN  [lsm303d] SPI init failed
ERROR [lsm303d] driver start failed
MPU6000 on SPI bus 1 at 4 (1000 KHz)
INFO  [mpu6000] accel cutoff set to 30.00 Hz
INFO  [mpu6000] gyro cutoff set to 80.00 Hz
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
WARN  [hmc5883] no device on bus 1 (type: 3)
INFO  [load_mon] stack check enabled
INFO  [px4io] default PWM output device
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
px4flow [149:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
No autostart ID found 
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> [KWARN  [logger] [logger] Not logging; SD almost full: 35 MiB

I tested that the hw type of pxihack v3 is v2M, and the V2M code seems to be closing some spi. I connect PB12 to 3.3v, hw type is v30, lsm303d and l3gd20 can work.

dagar, 2018-07-13:
As discussed, this is more or less an issue with the fmu-v2 board determination. https://github.com/PX4/Firmware/blob/master/src/drivers/boards/px4fmu-v2/init.c#L192-L264

Any suggestions for updating determin_hw_version to properly detect this?

davids5, 2018-08-14:
@cuhome please test https://github.com/PX4/Firmware/pull/10231

closed 2019-02-04 by stale[bot]